Candas1/Split_Hoverboard_SimpleFOC #1: please join and help us :-)

RoboDurden, 2023-06-15:
Hi @Candas1 , @robcazzaro i guess the candas fork should now be the place of activity and indeed the old https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/11 needs seconds to load on my mobile data plan.

> @RoboDurden Do you know if the emergency stop is really connected on this layout? If yes it will act as an overcurrent trigger and stop the pwm output. It would only be a matter of enabling [This](https://github.com/Candas1/Arduino-FOC/blob/6ed4e359df771d913d339e781af221427102e5ff/src/drivers/hardware_specific/gd32/gd32_mcu.cpp#L102) with the right polarity and would be safer.
> 
> [EDIT] And probably setting PB12 [properly](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/c350d800c7913a62c501d02a916795ebcdea70dc/HoverBoardGigaDevice/Src/setup.c#L192)

I can probe PB12 with a dso to see if it offers analog or digital input. But according to your spreadsheet 
https://docs.google.com/spreadsheets/d/15msbDAIMxC2rIkq8Au8vf82ub1qEaS67Lc1cFb86Jpc/edit#gid=0
"sed on some of the boards as overcurrent trigger when the dc current is high enough to turn a transistor ON. Should disable PW" 
i need to find that transistor and see if i can verify an analog input there while putting a hand on the motor with my gen2.x firmware running.

But according to the 2.0 schemtaics, PB12 is functional:
[image]

With the mosfet only driven by the shunt voltage, it should open slowely and indeed produce some analog value on PB12 when some strong current reaches the the gate threshold.

> This sketch is already using 91% of the flash, the GD32F130C6T6 have really little memory. I wasn't able to add the serial commander for example, it was exceeding the size.

I checked the photos of the various layouts. Only 2.1 and 2.5 have a C6 mcu. The others have a C8 with doubled flash and ram = 64 kB flash and 8 kB ram. (2.2 has K8 which only has less pins)

So to get all features running i think for now it is okay to compile with the C8 enviroment :-)

Candas1, 2023-06-15:
> Hi @Candas1 , @robcazzaro i guess the candas fork should now be the place of activity and indeed the old [RoboDurden/Hoverboard-Firmware-Hack-Gen2.x#11](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/11) needs seconds to load on my mobile data plan.
> 
> > @RoboDurden Do you know if the emergency stop is really connected on this layout? If yes it will act as an overcurrent trigger and stop the pwm output. It would only be a matter of enabling [This](https://github.com/Candas1/Arduino-FOC/blob/6ed4e359df771d913d339e781af221427102e5ff/src/drivers/hardware_specific/gd32/gd32_mcu.cpp#L102) with the right polarity and would be safer.
> > [EDIT] And probably setting PB12 [properly](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/c350d800c7913a62c501d02a916795ebcdea70dc/HoverBoardGigaDevice/Src/setup.c#L192)
> 
> I can probe PB12 with a dso to see if it offers analog or digital input. But according to your spreadsheet https://docs.google.com/spreadsheets/d/15msbDAIMxC2rIkq8Au8vf82ub1qEaS67Lc1cFb86Jpc/edit#gid=0 "sed on some of the boards as overcurrent trigger when the dc current is high enough to turn a transistor ON. Should disable PW" i need to find that transistor and see if i can verify an analog input there while putting a hand on the motor with my gen2.x firmware running.
> 
> But according to the [2.0 schemtaics](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/c350d800c7913a62c501d02a916795ebcdea70dc/Schematics_2.0/HoverBoard_CoolAndFun.pdf), PB12 is functional: [image]
> 
> With the mosfet only driven by the shunt voltage, it should open slowely and indeed produce some analog value on PB12 when some strong current reaches the the gate threshold.
> 
> > This sketch is already using 91% of the flash, the GD32F130C6T6 have really little memory. I wasn't able to add the serial commander for example, it was exceeding the size.
> 
> I checked the photos of the various layouts. Only 2.1 and 2.5 have a C6 mcu. The others have a C8 with doubled flash and ram = 64 kB flash and 8 kB ram. (2.2 has K8 which only has less pins)
> 
> So to get all features running i think for now it is okay to compile with the C8 enviroment :-)

OK I could try to set it up and we could test by pulling this pin high.
Base on your screenshot it would be useful only in case of short circuit/very high current, but it's still useful.

Candas1, 2023-06-15:
To do:
Note the zero_electric_angle after sensor alignment, and feed it with [initFOC](https://docs.simplefoc.com/bldcmotor#step-61-skip-alignment---position-sensor) to skip the alignment

I think the sensor.update in the loop is redundant, it's already in the [loopFOC](https://github.com/simplefoc/Arduino-FOC/blob/05954cb9691534dd478407505d810a68f1673a5e/src/BLDCMotor.cpp#L317)

RoboDurden, 2023-06-15:
Still tidying up. I have removed the i2c beginning (that you got when forking my repo) and uploaded the final i2c test code here: https://github.com/RoboDurden/GD32_I2C_Slave

Things like `zero_electric_angle after sensor alignment` are totally new to me as i only debugged the HallSensor. Will need some time to safely get your nice repo running.

With the simple motor mode i guess position control would be nice to have next.
Then adding I2CCommander would already allow to use multiple gen2 boards to build a robot arm. Something the gen1 firmwares are not made for. And FOC efficiency would not be that important for using the bldc as a servo motor.

@Candas1 , maybe try not to quote a full response to keep this issue as short as possible. It will grow vastly anyway :-)

Candas1, 2023-06-15:
To be honest, I didn't mean to fork your repository and take over.
I wanted to save time building up on what you already did.
My changes could also be pulled to your repository and you could continue from there.

RoboDurden, 2023-06-15:
It is an honor for me that you did fork my code witht the little CIO class and my debug macros.
I guess more low level gd32 specific code has to be debuged and updated so it might be best when the work continues here.

RoboDurden, 2023-06-15:
# Unfortunately, my first gen2.0 burned with your code :-/ 
## That can happen :-)
Maybe i should have enabled this emergency stop first ?
I stepped through the entire setup (F5, F11, F10) to see your code. When everything looked okay at the end of setup() i hit F5 to let the loop lose.
Then the motor made one step forward and one step backward and smoke started to come from somewhere on the board.

Now both mosfet of the green phase have a short over all three pins. and the low-side mosfet of the middle phase blue has a short from sorce to drain but gate still isolated.

Now i need to unsolder the mosfet and find replacements. With these analog gate drivers i can only test the low-side without a new mosfet built in because the high-side needs to low-side and the bldc connected to generate the +14V gate voltage for the high side.

I guess the driver class should have a protected mode that ensures that no low-side and high-side open at the same time ?
I fear that happened for the left green phase.
The burn of the middle low-side then i fear must have happend because the bldc phases were connected.

Maybe some kind of dry run would be nice were the 6 gate driver pins are mapped to something like the 5 leds ?

I did compile and run the C8 enviroment. But in the past it made no differece for the board to upload a c6 or c8 binary.

As i mentioned, i am embarking on a 1-2 week journey tomorrow, so i guess i will not proceed a lot today but start packing. 
Maybe i can replace the mosfet today or find some to do it on the journey.
Bit I still have a second board on my test setup :-)

RoboDurden, 2023-06-15:
I have not yet looked into the driver code. But it would be good if it had a save mode where it simply checks that no mosfet is turned on when the opposite low/high side is still on. Does the driver class blindly sets gates to high without checking that ?

Candas1, 2023-06-15:
It's not working like that.
You set a pwm duty cycle, complementary channels and deadtime, the MCU handles the rest.
It will never turn both ON at the same time.

I made sure this code runs with conservative values and is not triggering CC mode of my power supply.

RoboDurden, 2023-06-15:
I tested the board without the 2 mosfet of phase green but then both mosfet of blue burned. Don't know if your firmware was still active. i think i am done for today.
At this stage i would not want to simply test my second board. 
We should find some test setup where the 5 leds and tx from uart1 are used as the 6 gate driver pins..
And then some idea how to test with a single channel dso if there is an overlap of some gate outputs.

Is it possible to add a debug interrupt that logs digital reads the gate output while the pwm interrupt is doing its work ?

RoboDurden, 2023-06-15:
An yes, i should have used the little 3A constant courrent power supply i have here in my train station: https://www.aliexpress.com/item/4000549476335.html
I forgot you warned me to use a battery :-)

Candas1, 2023-06-15:
Yes that's the reason why.
But I am disappointed now, I will check what could be wrong.
Sorry about this.

RoboDurden, 2023-06-15:
I think I already found another 2.0, but the led headers are a different. The chip layout seems to be identical:
[image]

robcazzaro, 2023-06-15:
@RoboDurden when building the code in Candas repository, by mistake I used the wrong SimpleFOC and ended up using the STM32-specific code for SimpleFOC (GD32 elsewhere). Platformio takes some time to upload the libraries referenced by Platformio.ini and if you start compiling too quickly after downloading it, it might pick up the wrong library and use the STM32 code for SimpleFOC

A way to test which code is used, is to put a breakpoint in the file gd32_mcu.cpp.c, in the _configure6PWM() function and see if it gets called. Or adding a debug print in that function

The MOSFET pins are driven directly by the timer outputs and if the timer is properly programmed, it should never set both high at the same time. Unless there is a stray piece of code that redefines the GPIO pins along the way.

As soon as my boards arrive, I can safely test this as well, using a logic analyzer to ensure that the complementary pins are never high at the same time

RoboDurden, 2023-06-15:
As i did step into all new setup functions i did see the new gd32_mcu.cpp.c code from Candas1 :-)

My one board is done. after unsoldering the next pair, i can power the board with that simple CC power supply and the 5V and 3.3V cascade is working. But the MCU does not respond to the ST-Link anymore. So the the mcu on board seems to be broken.

But i have tested the second board with my gen2.x firmware and my Split_Hoverboard_SimpleFOC from which you did fork.
So i am ready to burn the next board. But that should happen after we have gotten some idea of why my first board was killed and find some save_mode without the true gates being driven - i suggest.

I am used to blow up things :-)

robcazzaro, 2023-06-15:
The safest way to understand what happened is to use a Bluepill with a GD32, attach a logic analyzer to the 6 outputs and set a trigger that triggers if the complementary outputs are both high. Then let it run in all conditions. If there is a stray GPIO code, it will catch it.

With a logic analyzer or scope, it's also possible to see if the deadtime is too small for the MOSFETs used. With an insufficient deadtime (and a battery capable of hundreds of amps), it's possible to overheat MOSFETs. Which is also why having a power supply with current limiter is a good idea, ad we have just been reminded :)

If the timer is properly programmed, the MOSFETs are "protected in hardware", so to speak, and no bug can cause issues. Alas, the way the code works, it's not possible to set a timer to work without actual outputs from GPIOs, so I can't see a way to test the code and prevent the GPIOs connected to the MOSFETs to be driven...

Candas1, 2023-06-15:
I added [here](https://docs.google.com/spreadsheets/d/15msbDAIMxC2rIkq8Au8vf82ub1qEaS67Lc1cFb86Jpc/edit#gid=795460288) some calculations, that can be useful.

The current deadtime in the gd32 driver(60 clock cycles) is actually same as what simpleFOC is using by default (2% of the PWM cycle). 1250ns is plenty. Eferu's deadtime is only 750ns.

In the past I had managed to blow all the mosfets of a splitboard in one go, I had set a target value in debug mode, the value was too high and there was no ramp.

But in this repository the target is really low. Still I need to add a ramp/filter.

robcazzaro, 2023-06-15:
@Candas1 looking at the [code](https://github.com/Candas1/Arduino-FOC/blob/6ed4e359df771d913d339e781af221427102e5ff/src/drivers/hardware_specific/gd32/gd32_mcu.cpp#L46), the deadtime set function is commented out. That's the one we discussed over email, using the STM32 macros to set the non-linear value in the register

So it looks as if at the moment there is no deadtime in the timer, which would explain the problems.... I think I sent you the snippet of code to replicate that STM32 function (but could not test it yet)

Candas1, 2023-06-15:
No no it's [there](https://github.com/Candas1/Arduino-FOC/blob/6ed4e359df771d913d339e781af221427102e5ff/src/drivers/hardware_specific/gd32/gd32_mcu.cpp#L101) but it's a macro. 
We need a formula later to use the parameter from simplefoc.

robcazzaro, 2023-06-15:
Oh, I see... you are using 60, which as you say it's 1250 ns. Would be worth double checking with an oscilloscope (I can do it, as soon as I have my boards)

The math is pretty complicated, given that it depends on the clock divider (0, 256 or 512), the CPU clock, and the register has a non-linear effect on the actual deadtime. I did send you the STM macro with all the needed definitions to make it work, let me find it and copy it here

Candas1, 2023-06-16:
Yes this is definitely in the todo list, the current value was just to get it to work, and it should be correct.
For someone to be able to set a different "dead_zone" using the simpleFOC library, we need to use the dead_zone parameter and convert it to the value the hardware needs.

Max from Arduino-gd32 was asking [here](https://github.com/CommunityGD32Cores/ArduinoCore-GD32/issues/95#issuecomment-1582272722) if there are expectable functions missing.
I feel like this function is missing in the GD32 SPL, not the arduino layer, but we can still mention about that.

marianomd, 2023-06-18:
Hi guys, regarding size optimization, I really encourage you to use [bloaty](https://github.com/google/bloaty). It's an outstanding tool.

Candas1, 2023-06-18:
Hi,

I tried debugging today, and even if I don't add any breakpoint and just run the code, my power supply definitely goes to CC mode. This is not happening when just flashing the firmware.

RoboDurden, 2023-06-18:
Great news :-)
Then I think I can try the second board without debug. my little step down LCD converter can only deliver 3A max and i will start with 1A max at 26V.
I should kill the second board anyway to make place for the next split board combo :-)

Candas1, 2023-06-18:
I did small changes to this repo and to the arduino-foc one.
You should pull it just in case as well.

robcazzaro, 2023-06-18:
Just a quick update: I received my Aliexpress order and managed to successfully do a brain transplant: I now have a working GD32F130C8 Bluepill. I have just checked that STLink can see and program it., not tried any actual code yet.

RoboDurden, 2023-06-18:
motor still stuttering and spinning very poorly after a few seconds: https://youtu.be/2OGnwagtY-U

It is the same stuttering that made my boad 1 burn after a stutter right followed by a stutter left !

@Candas1 , i did not fork your fork but cloned it with my Windows Github Desktop. 
I did remove the old I2C beginnings for which i by now have create a new repo. 
But when doing so i answered the prompt that i want to contribute to your repo.
How can i make a pull request without forking your repo ?

But you can quickly remove the i2c stuff in main.cpp yourself anyway :-)

After that video i tried all three combinations of 
```
//BLDCDriver6PWM driver = BLDCDriver6PWM( BLDC_BH_PIN,BLDC_BL_PIN,  BLDC_GH_PIN,BLDC_GL_PIN,  BLDC_YH_PIN,BLDC_YL_PIN );
//BLDCDriver6PWM driver = BLDCDriver6PWM( BLDC_BH_PIN,BLDC_BL_PIN,  BLDC_YH_PIN,BLDC_YL_PIN,  BLDC_GH_PIN,BLDC_GL_PIN );

//BLDCDriver6PWM driver = BLDCDriver6PWM( BLDC_GH_PIN,BLDC_GL_PIN,  BLDC_BH_PIN,BLDC_BL_PIN,  BLDC_YH_PIN,BLDC_YL_PIN );
//BLDCDriver6PWM driver = BLDCDriver6PWM( BLDC_GH_PIN,BLDC_GL_PIN,  BLDC_YH_PIN,BLDC_YL_PIN,  BLDC_BH_PIN,BLDC_BL_PIN );

//BLDCDriver6PWM driver = BLDCDriver6PWM( BLDC_YH_PIN,BLDC_YL_PIN,  BLDC_GH_PIN,BLDC_GL_PIN,  BLDC_BH_PIN,BLDC_BL_PIN );
BLDCDriver6PWM driver = BLDCDriver6PWM( BLDC_YH_PIN,BLDC_YL_PIN,  BLDC_BH_PIN,BLDC_BL_PIN,  BLDC_GH_PIN,BLDC_GL_PIN );
```
But no change in motor behavior. So it seems to be a timing issue ?

It is very convenient to have my Gen2.x firmware opened with Keil so i can simply hit F8 to very quickly stop the stuttering and reflash the 2.0_master_test binary (which does nothing when the onoff button is permanently switched on).
Also good to confirm that the hardware is still okay.

Now i wonder why your motor is not stuttering.
But does it also run as smootly as the normal block commutation of the gen2.x firmware ?

Greetings from Germany. @robcazzaro is in the California time zone. what is your time zone Candas ? And is Candas your first name ??

Candas1, 2023-06-18:
That's not stuttering, that's the sensor alignment.
If we feed initFOC with the values given by the sensor alignment, it won't happen anymore at startup.

My motor runs the same, I need to play around with the different modes.

Pull requests work only if you fork the repo.

I am from France, I am leaving close to Germany and Switzerland actually.

RoboDurden, 2023-06-18:
Maybe i or you should post my video to the simpleFOC community and someone with BLDCDriver6PWM will quickly identify the poor spinning ?
But if you have some ideas to test, i will happy wait for your next changes.
Good night and good luck :-)

RoboDurden, 2023-06-18:
in `setup()` you can also remove old uart2 test code:

```
  int iRX0old = PA10;
  int iTX0old = PA9;

  int iRX0new = PB7;
  int iTX0new = PB6;

  int iPinRX0 = RX0;
  int iPinTX0 = TX0;
```

You can also search and delete the two lines with `oPB6`

And the two uart return to sender stuff:

```
  #ifdef SERIALDEBUGXX
    if (SERIALDEBUG.available()) 
    {
      while (SERIALDEBUG.available() )
      {
        SERIALDEBUG.write(SERIALDEBUG.read());
      }
    }
  #endif
```
and
```

  #ifdef SERIALDEBUG
    if (SERIALDEBUG.available()) 
    {
      while (SERIALDEBUG.available() )
      {
        SERIALDEBUG.write(SERIALDEBUG.read());
      }
      SERIALDEBUG.println();
    }
  #endif
```

:-)

Candas1, 2023-06-18:
> Maybe i or you should post my video to the simpleFOC community and someone with BLDCDriver6PWM will quickly identify the poor spinning ? But if you have some ideas to test, i will happy wait for your next changes. Good night and good luck :-)
I shared my video [here](https://community.simplefoc.com/t/hoverboard-main-board-with-simplefoc/2610/37?u=candas1) already, I think it works exactly the same in your case.

I need to experiment more.
I suggested some improvements [here](https://community.simplefoc.com/t/hall-effect-sensor-skip-calibration/235/20?u=candas1), that would make sure the phases and halls are mapped correctly, but I would need time to implement this.

robcazzaro, 2023-06-18:
Unless I did something seriously wrong, I would recommend to stop using this code on a board connected to a power supply and with MOSFETs attached

In order to test the PWM code, I used the PWM6 standalone driver example (below). 

CH1 is connected to the H output, CH2 to the L output.

On the pins outputting 0, I see this. i.e. both MOSFETs ON at the same time (minus some spikes every ~38usec)

[image]

On the pins outputting 1 and 3 respectively, I see

[image]
[image]

As you can see, the same spikes every ~38usec, which cause the MOSFETs to conduct at the same time (every now and then). Also, the pin assignment seems wrong (G and B are inverted, so what should have been the 0 output was on the B pins instead, PA9 and PB14)

With the STM32 Bluepill and the same code, everything is as expected (0, 1 and 3 in order)
[image]
[image]
[image]

RoboDurden, 2023-06-19:
Thank you @robcazzaro very much. I first try to understand the stm32 outputs:

[image]

The upper image are hi-side (yellow) and lo-side (blue) mosfets conected to phase B,
middle image hi and lo connected to phase G
and lower image hi and lo to Y(ellow) ?

Then in about the first 30% of the full cycle i see a "negative" current flowing from phase G to B through the coils inside the motor (hi-side of B is on and lo-side of G is on).
Followed by 70% of the period a current from Y to G (hi-side of Y is on and lo-side of G is on)
and also the first 90% a current from Y to B (hi-side of Y is on and lo-side of B is on).

But how does that correlate to `driver.setPwm(1, 0, 3);` ? 

These three float values could also be negative, resulting in "opposite" current flow ?

As i have not read the simpleFOC code, for me it is still a black box :-/

Candas1, 2023-06-19:
[image]

That's what I see on the high side and low side of one phase.

Candas1, 2023-06-19:
I think [this](https://docs.simplefoc.com/bldcdriver6pwm#standalone-driver) is what @robcazzaro used.
So setting the pwm for each phase.

RoboDurden, 2023-06-19:
@candas1 , your photo is with the simple `driver.setPwm(1, 0, 3); test code ?
The blue pill does not yet have hall inputs to test the more complex code of this repo here.

Candas1, 2023-06-19:
> So 3.0/12.0 = 25% duty cycle on phase A, 6/12 = 50% on the middle phase cable And 5/12 on the C = yellow cable ? All aligned to the 20khz ?? I do not understand how these 3 floats can drive the bldc..

You provide a voltage for each phase, that simpleFOC tranlates to a duty cycle between 0 and 1.
The duty cycle is converted to a timer pulse by the driver.
High side/low side is handled by the MCU using the frequency and deadtime.

RoboDurden, 2023-06-19:
Thanks Candas, now I understand :-)
When the voltages are set to be 1,2,6 with 12V max, then all three phases are set to high the first 1/12 of the duty cycle and no current flows.
Then phase A is pull down and currents start to glow from phase b and c to a. 
After another 1/12*1/20000 second, phase b is also pulled to gnd. Therefore current from b to a had been 1/12 which is proportional to the voltages 1 and 2 set in setPwm :-)
Phase c continues for another 4/12 and 4 is exactly the voltage difference between 2 and 6 :-)
And current from C to A has flown for 5/12 of the duty cycle.which is exactly 6-1 .

Candas1, 2023-06-19:
Yes, you're setting a pwm duty cycle/voltage, but the important one is the line to line voltage, that's when the current flows.

If you are interested, I like this [intro](https://www.switchcraft.org/learning/2017/3/15/space-vector-pwm-intro).
We are not using SVWPM yet, only trapezoidal for now, but this will be relevant later.

Candas1, 2023-06-19:
The sensor alignment is weird, the motor would sometimes spin smoothly, sometimes poorly. 
I checked the zero_electrical_angle and direction in debug/monitoring and picked the values when it seemed to be smooth, and used it when initializing FOC.
It's available with my last [changes](https://github.com/Candas1/Split_Hoverboard_SimpleFOC/commit/e243d3a0ed4baa6ee9850dc474f11e31cf3b22ec).
I also tried trapezoidal 150 instead of 120 but I don't hear any difference.

robcazzaro, 2023-06-20:
Just a quick update here that I solved most of the timer/PWM issues and things should work better https://github.com/Candas1/Split_Hoverboard_SimpleFOC/issues/4#issuecomment-1597999614. But a PWM value of 0 (i.e. a phase temporarily at 0V) still results in a wrong output, so it's still unsafe

Candas1, 2023-06-20:
@robcazzaro, I shared my comments and will check this evening what can be improved.
I am leaving on Friday for 2 weeks and I still have to pack, so I will focus on those improvements this evening and will share an updated version.
Some of this issues where already there on the original firmware, but I agree we should be as close as possible to the STM32 behavior not to have bugs with simpleFOC later.

When I come back I will be able to experiment with those boards that are using a STM32F103:
[image]
I added the layout on the [spreadsheet](https://docs.google.com/spreadsheets/d/15msbDAIMxC2rIkq8Au8vf82ub1qEaS67Lc1cFb86Jpc/edit#gid=0) based on this [firmware](https://github.com/JohMah/Hoverboard-JC2015.7.31.V2.0-Hack). The pins for phases and halls are the same.
This will help compare between GD32 and STM32 boards behavior, and find differences on the arduino or simpleFOC driver implementation.

RoboDurden, 2023-06-21:
@Candas1 i do not really understand why you have removed the C6 enviroment as it should still fit in the 32 kB flash:

```
RAM:   [=====     ]  52.7% (used 4316 bytes from 8192 bytes)
Flash: [====      ]  44.4% (used 29116 bytes from 65536 bytes)
```

A usable Split_Hoverboard_SimpleFOC  seems to be in reach and as we have the pin definitions of  one C6 and one K6 board, i would like to do my best to support them.

The X6 mcu only has 4 general timers(16-bit) instead of 5. Is that the problem ??

P.S. If you have a grilfriend with you on holidays, you should put this simpleFOC project on halt for the next two weeks !

robcazzaro, 2023-06-21:
@RoboDurden, Platformio is weird and having a dual C8 C6 setup makes everything harder now. Once we have the C8 working, making a C6 version is going to be super-easy. No point in having extra complexity now