Candas1/Split_Hoverboard_SimpleFOC #3: Project dependency on https://github.com/Candas1/Arduino-FOC.git requires manual process to update

robcazzaro, 2023-06-18:
I just realized that when there is a checkin in https://github.com/Candas1/Arduino-FOC.git, it doesn't get automatically processed just by updating (pull) this project.

To make sure all the most recent changes in https://github.com/Candas1/Arduino-FOC.git are picked up, the safest way is to remove the .pio directory and let Platformio re-download the most recent code from github.

Otherwise it's possible to copy the changed files into .pio\build\libdeps\GD32F130C8\SimpleFOC\ and .pio\build\libdeps\GD32F130C8\SimpleFOC\

This is simply a side effect of not having released versions of the GD32 SimpleFOC  and the ability of updating them from the Platformio UI

Candas1, 2023-06-19:
Hi Rob,

I was using this [command](https://docs.platformio.org/en/latest/librarymanager/dependencies.html#updating-dependencies) to update arduino-gd32 after fixes have been published.
It can probably be used to update all dependencies.

robcazzaro, 2023-06-19:
It is possible to open the Platformio Core CLI and issue a command like `pio pkg update -l `

But it requires a library name. And when I list the installed packages for the project I get

```
PS C:\GD32\Candas> pio pkg list      
Resolving GD32F130C8 dependencies...
Platform gd32 @ 1.0.0+sha.3424e66 (required: git+https://github.com/CommunityGD32Cores/platform-gd32.git)
├── framework-arduinogd32 @ 4.20000.210603+sha.dfa7ccb (required: git+https://github.com/CommunityGD32Cores/ArduinoCore-GD32.git)
├── tool-dfuutil @ 1.9.200310 (required: platformio/tool-dfuutil @ ~1.9.190708)
├── tool-openocd-gd32 @ 2.1100.211207 (required: communitygd32cores/tool-openocd-gd32 @ ~2.1100.0)
├── tool-sreccat @ 1.164.0 (required: platformio/tool-sreccat @ ~1.164.0)
├── tool-stm32duino @ 1.0.2 (required: platformio/tool-stm32duino @ ~1.0.1)
└── toolchain-gccarmnoneeabi @ 1.90201.191206 (required: platformio/toolchain-gccarmnoneeabi @ ~1.90201.0)

Libraries
├── RTT Stream @ 1.4.0 (required: koendv/RTT Stream @ ^1.3.0)
│   └── ANSI @ 0.2.0 (required: ANSI)
└── Simple FOC @ 2.3.0+sha.7cff1fa (required: git+https://github.com/Candas1/Arduino-FOC.git)
```

For the "gd32" package, the command ```pio pkg update -g -p gd32``` works, because it recognizes gd32 as a platform name

Candas1, 2023-06-19:
I used it without a parameter and it worked

robcazzaro, 2023-06-19:
Yes, but I was worried about side effects potentially updating other Platformio projects. I have a few dozens Platformio projects for all processors with various version of libraries. Updating the wrong library would mess up things badly (sometimes a newer library is incompatible and breaks a 3 years old project)

with ```pio pkg update -l "Simple FOC"``` it's guaranteed to only update that single library

Candas1, 2023-06-19:
ok I thought it was only updating dependencies for the current project/environment.

Candas1, 2023-06-19:
Ah we can probably use [that](https://docs.platformio.org/en/stable/core/userguide/pkg/cmd_update.html#cmdoption-pio-pkg-update-no-save)

robcazzaro, 2023-06-20:
Good idea adding the Custom option!

Incidentally, for the time being I would remove the C6 environment. We are all using C8 and having both can cause confusion for some future new contributor. I also found it takes longer to update as it copies the libraries to multiple directories

Candas1, 2023-06-20:
I wanted to use the same layout as Robodurden is using for his tests, this layout has a C6 chip.

RoboDurden, 2023-06-20:
@Candas1 now i am a little bit confused myself :-) My test setup with cc power supply that i have with me in my solar camper is a gen2.0 board with c8 chip. The gen2.5 test setup with c6 is back at my train station.

Candas1, 2023-06-20:
Well I am either stupid or blind.
I have always thought it was a C6.
So I am wasting our time.
My fault sorry.