Candas1/Split_Hoverboard_SimpleFOC #5: motor does not reach max speed like old Gen2.x binary

RoboDurden, 2023-06-23:
> In the past, I had problems with SimpleFOC due to the limited speed of the encoder readings. The Arduino interrupt handling code has a lot of overhead, made worse by the various processor layers. If the code handling the hall sensors is not fast enough, speed is limited by that. And we are using a 48MHz processor, pretty slow all considered.
> 
> I don't have time today, though. Over the weekend, I'll look at ways to speed up the interrupt handler and see if that improves anything.

I might be easier to slow down the hall sensor reading artifically and see if that has an effect on the motor max speed :-)

Candas1, 2023-06-23:
I am still not sure if the sensor alignment is good.

One way to check this is described [here](https://e2e.ti.com/support/motor-drivers-group/motor-drivers/f/motor-drivers-forum/845535/faq-how-to-ensure-correct-alignment-of-hall-sensors-and-your-motor)

Do you see/hear differences in how the motor spins in each direction with both binaries? 
A wrong offset could lead to phase advance in one direction, and phase delay in the other direction.

RoboDurden, 2023-06-23:
you can hear in the video for yourself but no the sound is the same.
The current however is about 540 mA spinning "left" and about 580mA spinning "right".
So there indeed is some asymmetry

Good night ! i am tired now and i still want to finish the rewatch of https://en.wikipedia.org/wiki/The_Hunt_(2012_film)
Good dvd...

robcazzaro, 2023-06-24:
@RoboDurden can you please do a quick test

Add the following to the plaformio.ini file

```
build_unflags = -Os
build_flags = -Ofast
```

It compiles the binary with the highest level of optimizations possible, and if my theory is correct (slow interrupt), it should speed up the motor by a small but measurable amount.

Meanwhile I'm looking at the interrupt code

RoboDurden, 2023-06-24:
@robcazzaro , I would rather do that tomorrow when I have arrived at my solar warehouse. I have already packed the test setup in a box. And my notebook is occupied with 3d printing..

RoboDurden, 2023-06-24:
If the effect is small i will need UART rpm logging and I have yet to setup that again.

robcazzaro, 2023-06-24:
> @robcazzaro , I would rather do that tomorrow when I have arrived at my solar warehouse. I have already packed the test setup in a box. And my notebook is occupied with 3d printing..

Of course, I didn't mean "right now" 😉 whenever you have time, no hurry

I saw a few Android apps that can measure RPMs if you stick a white tape to the wheel, maybe that is easier than enabling UART. I mean, UART is good to have, but an RPM meter app might be a quick way to see if there are differences

robcazzaro, 2023-06-24:
@RoboDurden I understand the interrupt handling code pretty well now, and there are a few hacks that would speed things up. But it requires a few changes to a few files and possibly breaking compatibility with the official GD32 core.

If, whenever you have time, you can confirm that compiling with -Ofast (the platformio.ini changes above) makes a difference, I will send you the modified files. Otherwise there's no point in having you waste time with the changes.

RoboDurden, 2023-06-25:
Okay that were my changes:

```
;build_flags = -D __PIO_DONT_SET_CLOCK_SOURCE__
;	-D __SYSTEM_CLOCK_48M_PLL_IRC8M_DIV2=48000000
build_unflags = -Os
build_flags = -Ofast	
```

I did not hear a higher speed and android apps do not use camera but inernal IMU and you have to put your phone on the turntable..

But current was sligtly less. before 590mA and 550mA (left/right) and with your requested changes: 550mA and 500mA

So i recorded the audio: www.robosoft.de/forums/rpmsound.mp3
first you hear simpleFOC-old left/right, than your new ini file left/right, and last the gen2.x left/right
And i made a fft analysis from the constant part of all three left(?) sounds:
[image]

So yes, your requested ini changes result in a 476 Hz / 456 Hz = 4.3% speed increase,
but the gen2.x still has another 547 / 476 = 15% rpm increase on top of your changes.

This was a long sunday for me. Got 3.5 kW old solar panles for free:
[image]
[image]

Good night and good luck :-)

robcazzaro, 2023-06-25:
That is really good news @RoboDurden. We can confirm it's a IRQ overhead issue. I will send you a couple of file to try before my end of day (overwriting the existing one for now). Thanks for doing this, and clever use of the audio, well done!

Just FYI: the GD32 Arduino core has some weird behavior with the interrupts, I opened an issue in their repository to be sure https://github.com/CommunityGD32Cores/ArduinoCore-GD32/issues/105. 

TLDR:
The GD32 interrupt handler (at a hardware level) groups GPIO 0-1, GPIO 2-3 and GPIO4-15 as separate interrupt source. So the EXTI01 handler must check if GPIO0 or 1 fired. Same for EXTI23. The problem is that any interrupt for pins 4 to 15 fires only one interrupt, and the handler must scan all 12 possible sources to see which one fired. So the HALL sensor on PB11 takes a long time, and the one on PC14 even more. The easy hack is to only check for pin=11 or pin=14 in the EXTI4-15 handler (which makes the code much, much faster), but I wanted to fully understand the GD32 logic for multiple ports and pins before messing with it.

Also opened https://github.com/CommunityGD32Cores/ArduinoCore-GD32/issues/106 to see if we can speed up the IRQ handler without having to change the GD32 Arduino files

Bottom line: as long as we know what interrupts we need for GPIO pins change, we can easily speed things up, and I have working code. Just want to finish testing it

robcazzaro, 2023-06-25:
Ok, here are 3 files to replace whenever you have time @RoboDurden 

C:\Users\[yourname]\.platformio\packages\framework-arduinogd32\cores\arduino\gd32\gpio_interrupt.c
[project_dir]\.pio\libdeps\GD32F130C8\Simple FOC\src\sensors\HallSensor.h
[project_dir].\.pio\libdeps\GD32F130C8\Simple FOC\src\sensors\HallSensor.cpp

Please note that one is in the platformio GD32 platform directory, the other 2 in the project directory. And please check that the changes are compiled. Incidentally, the VS Code UI sometimes gets confused by all the nested defines, so in the gpio_interrupt.c file, the portion following `#elif defined(GD32F3x0) || defined(GD32F1x0)" is shown as disabled (greyed out), but it's active and gets compiled (GD32F1x0 is defined). It's a VS code UI bug. That one is the file I hope not to have to change thanks to [this ](https://github.com/CommunityGD32Cores/ArduinoCore-GD32/issues/106)

Please also keep the -Ofast optimization enabled for now. It will make debugging harder, but a 4% gain is non-trivial. There are further optimizations possible in the Hall sensor code (updateState()), but those require more thinking. The ones I did for now are pretty safe.

If anything is unclear, just ask. I tested everything and it seems to work as expected, but it would not be the first time I screw up 😊

[hall_IRQ.zip](https://github.com/Candas1/Split_Hoverboard_SimpleFOC/files/11861168/hall_IRQ.zip)

robcazzaro, 2023-06-26:
BTW: I have been looking at other ways to potentially speed things up, but would require changes here and there. If the interrupt enhancements are not enough, I have a few more things to try. But the GD32 at 48MHz is slower than most other processors recommended for SimpleFOC, so we might have to hand-optimize more. To be fair, the FOC algorithms are more compute-heavy than the type of control implemented by most hoverboards

Candas1, 2023-06-26:
Don't be pessimistic, I am sure we will get there.
We are not doing foc yet, but I assume that the boards with current sensing had foc implemented.

We for sure will have to optimize because of the arduino-gd32 and simplefoc overhead.

RoboDurden, 2023-06-26:
no, the new three file do not make a difference:
[image]

The gen2.x also has a speed difference with left/right, but only 0.7%.
the simpleFOC has 1.5%.

I think the gen2.x has no interrupts for the three hall sensors at all but simply reads the state in the bldc.c 16 kHz CalculateBLDC():

```
	// Read hall sensors
	hall_a = gpio_input_bit_get(HALL_A_PORT, HALL_A_PIN);
	hall_b = gpio_input_bit_get(HALL_B_PORT, HALL_B_PIN);
	hall_c = gpio_input_bit_get(HALL_C_PORT, HALL_C_PIN);
```

So i do not really think that the three hardware interrupts from HallSensor class take longer or are less accurate ?

The EFeru FOC is faster than the Gen2.x anyway and with field weakening it can get even faster (20%). So maybe we should procceed with current sensing to get simpleFOC foc working and then have new options to tune speed and power consumption.

Candas1, 2023-06-26:
Foc is actually slower than trapezoidal or sinepwm, even with EFeru's firmware.

robcazzaro, 2023-06-26:
Bummer. And thanks @RoboDurden for your additional tests.

Clearly, even if the interrupt code was inefficient, it did not consume enough cycles to be a meaningful difference. Was a low hanging fruit, but not the right one.

There might be other opportunities (for example the sensor.updateState() function is over complex and calls bloated functions (like micros() that is needlessly complex, would be easier to re-implement using DWT_CYCCNT and knowing that the core clock is 48MHz). @RoboDurden if you try following the hall sensor code, you will see that a simple 3-line GPIO read in the gen2.x software is more than 50 lines, called each hall interrupt (open and close). SimpleFOC abstraction makes things more complicated (SimpleFOC can for example use SPI encoders for angle control, and the same code works with hall sensors)

As for the speed difference, is 0.7%. We are running the GD32 suing the internal oscillator, which has a 1% accuracy, so that could explain the small difference you are seeing.

The next step unfortunately it's profiling the code. The best way is to use SWD/SWO, but if you have a cheap STLink Chinese clone, the SWO pin is not enabled. It can be enabled with https://lujji.github.io/blog/stlink-clone-trace/ and it's well worth it, but time consuming and you need to be good at soldering on the small STM32 pins. With that enabled, you can ten use the STM32CubeIDE from ST to run the SWV Statistical Profiling (it's under Window, Show View, Other
[image]

That would get you a complete profiling profile, but might be too slow over SWO and pretty hard to get running.

Otherwise use macros and DWT_CYCCNT to measure execution cycles by function, which is probably the best way to go for us here. There are even available implementations like https://github.com/Serj-Bashlayev/STM32_Profiler. DWT_CYCCNT is a standard M3 core register, so available on the GD32. That library prints using SWO (once again), but it could also print to a debug output after a fixed time profiling.

Or it would be easy to enable DWT_CYCCNT  once at the beginning, create a structure for the various functions we use, store the DWT value upon entering the function, subtract the value at the end from the start, and add the instruction count to the right element of the structure for that function

```
//the following code needs to run only once when enabling the profiling

// enable DWT
CoreDebug->DEMCR |= 0x01000000;
// Reset cycle counter
DWT->CYCCNT = 0;
// enable cycle counter
DWT->CTRL |= 0x1;
```
```
// for each function
uint32_t start = DWT->CYCCNT;

// code to be measured here

structure.this_element += DWT->CYCCNT-start;
```

Then stop the code after, say, 1000 loops and print the values in the structure

Alas, I can't profile on the Bluepill dev board. Without a motor and hall sensors, the code won't work as expected. If you think you can run the profiler, I'd be happy to then analyze the code and see where we can gain. If you can't, I'll have to set up my only board and hope for the best

RoboDurden, 2023-06-26:
Well all the many lines you wrote here about such a profiler does not make me wanting to do it at all :-/
My dso single channel did show that the simpleFOC and gen2.x waveforms are quite alike.
So.why not find a clever way to see what is different...
I think candas mentioned a different alignment.
I think I have a cheap multi channel logic analyser in my train station but I forgot to take it with me to my solar warehouse.
Logging the three hall inputs and some gate outputs might shed some light in what simpleFOC is doing differently or badly.

If there are some delay parameters we could program an Auto-Tuning that optimizes rpm with a steepest descent over all parameters..

I would prefer some direct approach instead of blindly increase analyzing..

Good night for today's

robcazzaro, 2023-06-26:
Profiling is not blindly increasing analyzing 😉. It's looking for inefficient areas of the code that are holding back performance. If we are performance limited now, it's only going to get worse once we add the more complex elements of the SimpleFOC code. Insufficient processor performance is one of the most common causes of problems reported in the SimpleFOC forums.

You can't examine external signals to see where the problem is. Let's assume we find that the PWM waveform is 3% delayed compared to other code. How would we go about changing that, if we don't know what's causing it?

The main take away from my previous comment is that measuring performance is as easy as adding 2 instructions, one at the beginning, one before the exit of each function, plus half a dozen at the end of setup(). If we use macros for the beginning/end of each function, we can enable/disable perf measurement during development. And that will benefit everything we do and help optimize in the future.

That is the clever way to see what's happening, IMHO. But I'd be happy to be proven wrong, if you can find something else that works. And, yes, it might be just a hall sensor alignment, that would be great. I'll hold on doing anything wrt profiling until you fully evaluate that option.

Candas1, 2023-06-27:
Just a reminder.
Gen 2.x is running at 72Mhz, the simpleFOC one at 48Mhz.

RoboDurden, 2023-06-27:
okay, max speed is even slightly higher (549 Hz left and 549 Hz right) then Gen2.x when setting these to init values to 20 instead of 26 (Volt): 

```
  driver.voltage_power_supply = 20;
  driver.voltage_limit = 20;
```

But the current high rises to 850-900 mA instead of the Gen2.x with less than 500 mA  :-/
I have the feeling that the problem is not code optimization but wrong pwm alignment and therfore inefficient motor control.
But with my 1 channel dso i fear i can not dig deeper.

RoboDurden, 2023-06-27:
short note to @Candas1 , that add_RTT_task.py needs
`    "\"$PROJECT_PACKAGES_DIR/tool-openocd/bin/openocd\"" + 
`
to handle path names with " " spaces.

Nice to have log output without another cable. But the sensor.getVelocity() output is not smooth at all:

```
GD32: 26500     0: 0    1: 1    2: 1    angle: 103.81   speed: 41.65
GD32: 26600     0: 1    1: 0    2: 0    angle: 107.79   speed: 38.98
GD32: 26700     0: 0    1: 1    2: 0    angle: 111.70   speed: 41.68
GD32: 26800     0: 0    1: 0    2: 1    angle: 115.61   speed: 35.01
GD32: 26900     0: 1    1: 1    2: 0    angle: 119.59   speed: 40.85
GD32: 27000     0: 0    1: 1    2: 0    angle: 123.43   speed: 41.02
GD32: 27100     0: 1    1: 0    2: 1    angle: 127.41   speed: 36.84
GD32: 27200     0: 0    1: 1    2: 0    angle: 131.39   speed: 41.31
GD32: 27300     0: 0    1: 1    2: 1    angle: 135.23   speed: 41.83```

Do not think that the rpm at max speed changes by 40.85/35.01 = 17% within 100ms

Maybe we should indeed implement/derive our own HallSensor class :-/

robcazzaro, 2023-06-27:
@RoboDurden the sensor is not filtered in SimpleFOC and very "spiky". There is a new smoothedsensor class being developed for that reason.

The folks running the forum recommend using motor.shaftVelocity() instead of sensor.getVelocity(). That is filtered and smoothed

As for your question on the clocks, the GD32 (like STM32) have multiple buses and clocks, each with PLLs and dividers, and it's a puzzle trying to understand how fast each bus runs. for STM32, there is a function in the STM32CubeIDE that helps calculate each and every PLL and divisor and gives you an idea in MHz or kHz. I haven't checked, but if Gen2.x is running at 72MHz, all bets are off with clocks and timing accuracy and I doubt it's. But if you want to truly understand the actual values, you need to look at the clock initialization and note each and every value being configured, then you can know the actual speed of a specific bus or component (in some cases, the speed can be half of the clock, depending on the edge polarity). Just to give you a sense, here's the clock tree for the GD32F130 (page 79, User manual)
[image]

robcazzaro, 2023-06-27:
@RoboDurden one more comment re:

```
void loop()
{
  // main FOC algorithm function
  // the faster you run this function the better
  // Arduino UNO loop  ~1kHz
  // Bluepill loop ~10kHz 
  motor.loopFOC();
  delay(1);
```
Adding a delay(1) is really a lot in what should be a tight loop. A loop() with only a single instruction of delay(1) runs at less than 1kHz, which is way too low for SimpleFOC. So adding a huge delay of 1msec in that loop will never work. For a more relevant test you might want to use delayMicroseconds(1); If that causes problems, that means that the processor is not fast enough with the current code. If even something like delayMicroseconds(10) causes no problems, that means that the code is fast enough and the problem is not code efficiency (but settings and hall alignment)

The goal is to run the loop at at least 10kHz to have a smooth performance. Alas, SimpleFOC is not timer-driven, but runs in a tight loop which is a bad way to write time-critical code.

From the Gen2.x code I can't understand how the GD32 clock is initialized. Usually there is a call to SystemClock_Config() to initialize the GD32 clock, but the Gen2.x code uses a non standard way.

```
	//SystemClock_Config();
  SystemCoreClockUpdate();
```
I don't have Keil, so can't quickly test. I could port the code to VSCode, but it will take some time. 

@RoboDurden If you have spare time, could you print the value of SystemCoreClock ? That variable is the GD32 clock (48000000 or 72000000). Knowing it will help making comparisons to the gen2.x code

Candas1, 2023-06-27:
We are free to move loopfoc and move functions to interrupts if we want to

https://community.simplefoc.com/t/simplefoc-theory-timing/2310/2?u=candas1

robcazzaro, 2023-06-27:
Good point @Candas1. Just a note that if we do that, we need to change how the hall sensor code works and disable those interrupts. Basically read the hall sensor upon receiving the timer interrupt like Gen2.x does (which would also make the velocity calculation much simpler and faster compared to the current one using the time between changes to calculate velocity, which as Robodurden noticed is very spiky). And make sure that the ADC code doesn't rely on interrupts either (Gen2.x uses ADC interrupts, but there are ways to start the ADC without interrupts). And look at how USART uses the interrupts in the Arduino code

Otherwise we have to deal with re-entrant interrupts and that causes all sorts of problems.

Candas1, 2023-06-27:
Yes we will trigger inserted adc at timer update event.

Candas1, 2023-06-27:
Some good advices were shared just now
https://community.simplefoc.com/t/hall-sensor-bldc-motor-does-not-turn/3046/10?u=candas1

RoboDurden, 2023-06-28:
According to the manual the GD32F13x is a 72 Mhz chip:
[image]

robcazzaro, 2023-06-28:
@RoboDurden not sure where you found that, but the datasheet has only 48MHz for the GD32F130xx https://www.gigadevice.com.cn/Public/Uploads/uploadfile/files/20230314/GD32F130xxDatasheetRev3.7.pdf and in this one https://www.gigadevice.com.cn/Public/Uploads/uploadfile/files/20230209/GD32F1x0_User_Manual_EN_Rev3.6.pdf it says `Note:(1) The maximum operating frequency of GD32F150xx is 72 MHz, and the maximum
operating frequency of GD32F130xx is 48 MHz.`

It looks like 72MHz was corrected in a revision `3.2 1. Modify 72MHz system frequency to 42MHz. Jul.25, 2019`. The GD32F150 runs at 72MHz, so maybe that's where the confusion comes from. It's entirely possible that most of the times it will work at 72MHz, but given how critical the correct working of the firmware is, I would not like to take that risk

RoboDurden, 2023-06-28:
[GD32F130K6U6_C81552.pdf](https://github.com/Candas1/Split_Hoverboard_SimpleFOC/files/11898769/GD32F130K6U6_C81552.pdf)

robcazzaro, 2023-06-28:
Well, as you can see, that is an old revision from 2018. After 2019 all versions have the 48MHz clock

RoboDurden, 2023-06-28:
@Candas1 i will delete my original `Split_Hoverboard_SimpleFOC `repo so i can fork your repo, add the `LowsideCurrentSense `code and make a pull request. Then @robcazzaro if he wish can start with a `SimpleFOC/src/current_sense/hardware_specific/gd32/gd32_mcu.cpp`. If not, i would try to port the Gen2 analog read of 
```
typedef struct
{
  uint16_t v_batt;  
  uint16_t current_dc;
} adc_buf_t;
```

by adding the two low-side currents like 
```
typedef struct
{
  uint16_t v_batt;  
  uint16_t current_dc;
  uint16_t current_B;  
  uint16_t current_C;  
} adc_buf_t;
```

As the Gen2 code does read v_batt and current_dc at 16 kHz (or 32 kHz?), i think that should also be good for the `LowsideCurrentSense` class.

Candas1, 2023-06-28:
I think we need to do something different.
We would sample currents with inserted adc, and read anything else separately. We don't need battery voltage 16000 times a second, it will only delay the current sampling and foc calculation.

RoboDurden, 2023-06-28:
google gives me no hits for "inserted adc" or "inserted adc stm32" or "inserted adc gd32" :-/
That Gen2-FOC-port you host also adds the low-side currents to the one adc_buffer:
```

typedef struct
{
  uint16_t v_batt;  
  uint16_t current_dc;
  uint16_t phase_a;  
  uint16_t phase_b;
  uint16_t mcu_temp;
} adc_buf_t;
```
https://github.com/Candas1/Hoverboard-Firmware-Hack-Gen2/blob/master/HoverBoardGigaDevice/Inc/defines.h#L49C1-L57C1

Here the GD32 init code: https://github.com/Candas1/Hoverboard-Firmware-Hack-Gen2/blob/master/HoverBoardGigaDevice/Src/setup.c#L368

With simpleFOC i think, the current_sense is synced to the motor which is simply triggered in the loop() :-(
What might work is to not  `motor.linkCurrentSense(&current_sense);` at all and instead do the `motor.loopFOC();` from our `LowsideCurrentSense `implementation when the ADC scan sequence is finished and `DMA_Channel0_IRQHandler(void)` gets called :-)

That would be my approach without knowing what "inserted adc" is.

robcazzaro, 2023-06-28:
@RoboDurden Gigadevice calls it inserted ADC, STM calls it injected ADC. It's a way to have multiple channels start ADC when an event (e.g. a timer) triggers it, without software assistance. SImpleFOC uses it to ensure that the current sensing is always time-aligned with the PWM. Adding interrupt servicing to the code will make it non deterministic (compared to a purely HW-based solution like injected ADC).

Here's a relevant snippet from the SimpleFOC [forum](https://community.simplefoc.com/t/concerns-and-suggestions-regarding-current-sensing-control-especially-in-b-g431b-esc1/2044/6?u=robca)

```All of our stm32 based low-side sensing is based on the ADC injected channels and TRGO events that are triggering the ADC conversion exactly in the middle of the time the low-side mosfets are active (which is effectively the start of the center aligned duty cycle). This is not the most optimal synchronization technique, I think that the STM motor suite uses the fourth timer channel to trigger the ADC which will usually result in a bit more precise measurements. But at this point we decided not to use this option because we do not want to use additional timer channel for the ADC sync as the user might prefer to use it for something else. But this might change in future.

So at the moment, to avoid potential problematic situations due to the ADC conversion lasting to long and entering the high-side activation of certain mosfets we suggest using the PWM frequencies that do not exceed 20kHz. This in most cases results in behavior.
```

I'll leave it to you two to decide the best way to structure the high level code, you both have a lot more experience with motor control. I'll be ready to help on anything once the main structure is in place.

Candas1, 2023-06-29:
Injected adc is also what the simpleFOC current sense driver is using. 
I have experience with injected adc and know what to do, so please leave it to me, I will check when I am back.
I think there is a lot more in the to-do list. 

And let's discuss adc in a new thread. 

Are we done with this thread? 
Now we are able to go as fast as gen2.X,  but not at maximum target value?

RoboDurden, 2023-06-29:
max speed like Gen2 is reached but at the cost of 850-900 mA instead of the Gen2.x with less than 500 mA :-/
If the Gen2 code has a motor efficiency of 80%, then with simpleFOC we have 42% :-(
I wonder how max speed (with no load) can be reached with such a low motor efficiency.
But when moving from the NiklasFauth (that Gen2 is based on) to EFeruFOC it was like having new motors in my solar vehicle: less sound, more torque, less current (indeed i never tested for max speed).
So maybe Gen2 is below 80% efficieny.

So no, we are not really able to go as fast as gen2.

P.S. i deleted my original Split_Hoverboard_SimpleFOC, did fork your version here, cloned my fork to Github-Desktop, made some code improvements and added current_sensor code and now Github-Desktop hangs and i can not push my changes to my fork (in order to make a pull request).

I guess deleting my original repo to replace it with your fork of my original now causes an infinite loop :-(
Ideas welcome :-)

Candas1, 2023-06-29:
I am still betting on sensor alignment.
Current sensing/foc won't solve this issue.
Current sensing/foc won't be beneficial without the precise angle the next release will bring with sensor smoothing.

RoboDurden, 2023-06-29:
I wrongly thought that current sensing somehow replaces the hall sensor angle and gives more precise angle ?
But the lowside current sensing will be worth nothing without a precise angle derived from a hall sensor smoothing ?

With the current HallSensor and it's three channel callbacks, the hall position is immediately updated i guess - but the pwm calculating motor.loopFOC(); is not called peridically but from every 112 ms up to 1122 ms :-(
I fear that  sensor alignment can not really work that way and we better sync Hall + Current + Pwm to one 16 kHz timer.

But you say that with hall sensor smoothing, the motor.loopFOC(); will get more precise angle and it only matters that the loopFOC is called often enough.

Then hall sensor alignment (to phase pwm) is not a problem because only one precise angle is important ?
Good to hear that the simpleFOC community is already working on such a hall sensor smoothing.

But current-sensor to phase pwm alignment will be very important, so we could continue with issue #6

Candas1, 2023-06-29:
Without smoothing, simplefoc only has one angle per hall position.
With smoothing, the angle will be extrapolated between 2 hall positions.
FOC needs that angle, and phase currents.

Smoothing wasn't available before because simplefoc was mainly used with encoders, they are much more precise than hall sensors already. 
I think the hall sensor support in simplefoc is not very old, so there is for sure room for improvement and contributions.

Sensor alignment initialization doesn't seem very accurate so we need some manual finetuning. This is not happening in the loop, and just gives you a zero electrical angle offset.

Candas1, 2023-06-29:
A motor is an electric magnet, using the 3 phases, you generate a 360 degree magnetic angle. This angles doesn't match with angle you get from hall sensor.

RoboDurden, 2023-06-29:
So you say that the angle doesn't match (you call it "not aligned") because the hall sensor is discrete and we need a smoothed hall sensor output. Then such a precise angle will really describe the current motor position and the motor.loopFOC will work better.
So you do not mean an alignment of digital rising/falling edges but simply a more precise motor position.
That has nothing to do with timers or interrupts :-)

But then still i wonder why Gen2.x is doing it so much better ???
Gen2 is not even updating the hall inputs with rising edge interrupts but simply samples them and does not smooth them either.

Candas1, 2023-06-29:
Your question was about the alignment, the precise angle is another topic.
Forget it, there is a lot of litterature about that, I can't summarize it all here.

The gen2.x firmware is only for hoverboards so it's already tuned.

RoboDurden, 2023-06-29:
btw, i introduced the BLDC_POLE_PAIRS   15 only after a short google search: "Standard 6.5 inch hoverboard hub motors have 30 permanent magnet poles, and thus 15 pole pairs.". Hope that is correct.

RoboDurden, 2023-06-29:
Okay i quickly added a linear interpolation which should also have an effect (on max speed) if sensor smooth would be the problem -  yes i see a higher max speed

4° coarse angle: 	459 Hz / 471 Hz
with linear interpolation	477 Hz /  486 Hz

That is a 3.5% increase in max speed. Current consumption is also about 50m less and quite like the Gen2.
But Gen2 reaches 547 Hz at the same current consumption.

```
199: 538        angle: -16.48   speed: 0.00     51325: -0.07
215: 501        angle: -15.50   speed: 5.36     13013: 0.07
200: 1215       angle: -10.40   speed: 0.00     4477: 0.07
216: 497        angle: -1.12    speed: 23.55    2965: 0.07
215: 1215       angle: 12.01    speed: 31.04    2249: 0.07
213: 543        angle: 28.00    speed: 30.01    2326: 0.07
227: 526        angle: 44.12    speed: 33.99    2054: 0.07
227: 547        angle: 60.18    speed: 31.40    2223: 0.07
228: 545        angle: 76.24    speed: 32.53    2146: 0.07
228: 1228       angle: 92.29    speed: 34.16    2044: 0.07
227: 549        angle: 108.49   speed: 35.21    1983: 0.07
229: 1226       angle: 124.41   speed: 30.82    2265: 0.07
228: 562        angle: 140.60   speed: 35.05    1992: 0.07
227: 544        angle: 156.66   speed: 32.44    2152: 0.07
228: 1228       angle: 172.86   speed: 31.25    2234: 0.07
226: 550        angle: 188.91   speed: 33.06    2112: 0.07
216: 540        angle: 204.90   speed: 29.80    2343: 0.07
215: 532        angle: 218.59   speed: 0.00     2732: 0.07
201: 1217       angle: 228.57   speed: 15.07    4634: 0.07
216: 532        angle: 234.36   speed: 7.86     8882: 0.07
201: 538        angle: 235.83   speed: 0.00     41090: 0.07
215: 486        angle: 234.92   speed: -6.10    11438: -0.07
217: 538        angle: 229.76   speed: -15.66   4459: -0.07
213: 1228       angle: 220.40   speed: -21.47   3252: -0.07
217: 1228       angle: 207.21   speed: -28.45   2454: -0.07
```
The first numbers of the last colum are the micro seconds between each change of the hall sensors. Even at max speed, there is a 2 ms gap that gets filled with my linear interpolation. And as the motor speed changes slowly, the linear inerpolation should already give a far more precise angle then the coarse 0.07 deg steps. 

The 0.07 degree is 2 * pi / (4°/360°). and as our hoverboard motors generate 90 ticks per revolution, that is 4° per hallsensor callback :-)

Here my simple linear interpolation:
```

float HallSensor::getMechanicalAngle() 
{
  long iTimeSinceLastUpdate = _micros() - aiTimeLastUpdate[0];
  float fGradient1 = (float)(aiAngleLastUpdate[0]-aiAngleLastUpdate[1]) / (aiTimeLastUpdate[0]-aiTimeLastUpdate[1]);
  long fAngleSteps = aiAngleLastUpdate[0] + (long)(fGradient1 * iTimeSinceLastUpdate);
  return ((float)(fAngleSteps % cpr) / (float)cpr) * _2PI ;

  return ((float)((electric_rotations * 6 + electric_sector) % cpr) / (float)cpr) * _2PI ;
}
```

Maybe i could now "fine tune" by adding a 0.00x angle offset so the `motor.loopFOC()` -> `setPhaseVoltage()` -> `driver->setPwm(Ua, Ub, Uc)` sees a slightly advanced motor angle.. But i think their FOC algorithms should be exact and not need such a tweaking

I am not that eager to advance from my linear interpolation to a more general https://de.wikipedia.org/wiki/Spline-Interpolation

RoboDurden, 2023-06-29:
Yes, asking Bing.chatGPT `"Schreibe mir in der Programmiersprache C den vollständigen Code für eine Polynominterpolation nach dem Horner-Schema welches auch die zugehörigen Koeffizienten berechnet."` gives a complete code example that might work:

```
// Dies ist ein Programm, das eine Polynominterpolation nach dem Horner-Schema durchführt und die zugehörigen Koeffizienten berechnet.
// Es verwendet die Funktion horner, die ein Polynom vom Grad n, einen Vektor von n+1 Koeffizienten und einen Wert x annimmt und den Wert des Polynoms an der Stelle x zurückgibt.
// Es verwendet auch die Funktion coeff, die eine Matrix von n+1 Punkten (x,y) annimmt und einen Vektor von n+1 Koeffizienten zurückgibt, die das interpolierende Polynom bestimmen.

#include <stdio.h>
#include <stdlib.h>

// Die Funktion horner berechnet den Wert eines Polynoms an einer Stelle x nach dem Horner-Schema
double horner(int n, double *a, double x) {
  double p = a[n]; // Initialisiere p mit dem höchsten Koeffizienten
  for (int i = n-1; i >= 0; i--) { // Iteriere von n-1 bis 0
    p = p * x + a[i]; // Multipliziere p mit x und addiere den nächsten Koeffizienten
  }
  return p; // Gib den Wert von p zurück
}

// Die Funktion coeff berechnet die Koeffizienten eines interpolierenden Polynoms für eine gegebene Matrix von Punkten
double *coeff(int n, double **points) {
  double *a = malloc((n+1) * sizeof(double)); // Alloziere Speicher für den Vektor a
  for (int i = 0; i <= n; i++) { // Iteriere von 0 bis n
    a[i] = points[i][1]; // Setze a[i] gleich dem y-Wert des i-ten Punktes
    for (int j = i-1; j >= 0; j--) { // Iteriere von i-1 bis 0
      a[j] = (a[j+1] - a[j]) / (points[i][0] - points[j][0]); // Berechne den Dividierten Differenzenquotienten für a[j]
    }
  }
  return a; // Gib den Vektor a zurück
}

// Die Hauptfunktion liest die Anzahl der Punkte, die Punkte selbst und den Wert x ein und gibt den Wert des interpolierenden Polynoms an der Stelle x aus
int main() {
  int n; // Die Anzahl der Punkte
  double **points; // Die Matrix der Punkte
  double x; // Der Wert x
  double y; // Der Wert y

  printf("Geben Sie die Anzahl der Punkte ein: ");
  scanf("%d", &n); // Lese n ein

  points = malloc((n+1) * sizeof(double *)); // Alloziere Speicher für die Matrix points
  for (int i = 0; i <= n; i++) {
    points[i] = malloc(2 * sizeof(double)); // Alloziere Speicher für jede Zeile von points
    printf("Geben Sie den %d. Punkt ein (x y): ", i+1);
    scanf("%lf %lf", &points[i][0], &points[i][1]); // Lese die x- und y-Werte des i-ten Punktes ein
  }

  printf("Geben Sie den Wert von x ein: ");
  scanf("%lf", &x); // Lese x ein

  double *a = coeff(n, points); // Berechne die Koeffizienten des interpolierenden Polynoms

  y = horner(n, a, x); // Berechne den Wert des interpolierenden Polynoms an der Stelle x

  printf("Der Wert des interpolierenden Polynoms an der Stelle %lf ist %lf\n", x, y); // Gib das Ergebnis aus

  free(a); // Gebe den Speicher für a frei
  for (int i = 0; i <= n; i++) {
    free(points[i]); // Gebe den Speicher für jede Zeile von points frei
  }
  free(points); // Gebe den Speicher für points frei

  return 0;
}
```
Sorry, i was not able to write the corresponding english prompt. This did not work:

"Write complete code for a polynomial evaluation according to the Horner's method in the C programming language, which also calculates the associated coefficients."

robcazzaro, 2023-06-29:
@RoboDurden please also keep in mind that SimpleFOC has already implemented a smoothed hall sensor class, discussion here https://community.simplefoc.com/t/smoothingsensor-experimental-sensor-angle-extrapoltion/3105/4 (I think @Candas1 also mentioned it before)

The code is in their tree, in the dev branch, this is the relevant check-in https://github.com/simplefoc/Arduino-FOC/commit/510527ae56fb70c6cf534e0a156e3c3c1f149046. It's not been released yet, but we could start using it and give feedback

RoboDurden, 2023-06-30:
!! Eureka !! - i have found it - after replacing the integer `mod %` with `fmod`, my simple linear interpolation works - exceptionaly good:

```
old HallSensor.cpp		460 Hz / 454 Hz at 650 mA
with 30% linear interplation:  	525 Hz / 529 Hz at only 380 mA
with 50% linear interplation:  	565 Hz / 561 Hz at only 400 mA :-)
with 60% linear interplation:  	579 Hz / 491 Hz at 480 mA :-(
Gen2				543 Hz / 544 Hz at 460 mA
```

Yes, the motor sometimes got stuck when changing rotation direction and makes an aweful sound :-(
But at low speed it is okay to simply disable the linear interpolation, and then that bad behavior disappeared :-)

I fear that the simpleFOC community only went for smoothing with a low pass filter. That is no prediction ..
My linear prediction might be to bold and only adding 0.5 of the linear gradient seems to be best.
Applying it at 100% will not make the motor spin correctly :-/

I still get some "ticks" at full speed that also happen on rotation changes. I guess when my prediction was to advanced and the next hall sensor event corrects this.

So a quadratic interpolation might be even better. Or indeed adding some simple `0.6 * fOld + 0.4 * fNew` low pass..

The chatGPT code seems to get stuck in an infinite loop.. will still be fun to analyze that.

```
float HallSensor::getMechanicalAngle() 
{
...
  if ((aiAngleLastUpdate[0]-aiAngleLastUpdate[1]) * (aiAngleLastUpdate[1]-aiAngleLastUpdate[2]) == 1) // no direction change
  {
    iTimeSinceLastUpdate = _micros() - aiTimeLastUpdate[0];
    fGradient1 = (float)(aiAngleLastUpdate[0]-aiAngleLastUpdate[1]) / (float)(aiTimeLastUpdate[0]-aiTimeLastUpdate[1]);
    float fAngleSteps = aiAngleLastUpdate[0] + (fGradient1 * iTimeSinceLastUpdate * 0.5);
    fAngleLin = (fmod(fAngleSteps,cpr) / (float)cpr) * _2PI ;
    return fAngleLin;
  }
  fAngleOrg = ((float)((electric_rotations * 6 + electric_sector) % cpr) / (float)cpr) * _2PI ;
  return fAngleOrg;
}
```

It really looks as good interpolation was the problem, 
and we now have a better motor then with Gen2, Alleluja :-)

RoboDurden, 2023-06-30:
Well this FOC code is still a black box to me:
```
void BLDCMotor::setPhaseVoltage(float Uq, float Ud, float angle_el) 
{
...
    case FOCModulationType::SinePWM :
      // Sinusoidal PWM modulation
      // Inverse Park + Clarke transformation

      // angle normalization in between 0 and 2pi
      // only necessary if using _sin and _cos - approximation functions
      angle_el = _normalizeAngle(angle_el);
      _ca = _cos(angle_el);
      _sa = _sin(angle_el);
      // Inverse park transform
      Ualpha =  _ca * Ud - _sa * Uq;  // -sin(angle) * Uq;
      Ubeta =  _sa * Ud + _ca * Uq;    //  cos(angle) * Uq;

      // center = modulation_centered ? (driver->voltage_limit)/2 : Uq;
      center = driver->voltage_limit/2;
      // Clarke transform
      Ua = Ualpha + center;
      Ub = -0.5f * Ualpha  + _SQRT3_2 * Ubeta + center;
      Uc = -0.5f * Ualpha - _SQRT3_2 * Ubeta + center;

      if (!modulation_centered) {
        float Umin = min(Ua, min(Ub, Uc));
        Ua -= Umin;
        Ub -= Umin;
        Uc -= Umin;
      }
      break;
...
  // set the voltages in driver
  driver->setPwm(Ua, Ub, Uc);
}
```
But i guess it is very sensitive to the motor angle, whereas Gen2 directly and non-tunable "block commutates" the three phases according to the hall sensors - not much to be done wrong.

I still have not undestood why setting `driver.voltage_power_supply` lower as the true battery voltage makes the motor even spin faster. And now with true voltage already exceeding Gen2, setting

`  driver.voltage_power_supply = 2.6 * BAT_CELLS; // power supply voltage [V]
`
will boost the max speed from `42.49`  to `55.28` and the current still is only 550 mA.
That is a 30% speed boost !!!
The Audacity fft frequency analisys no longer works because the sound spectrum has changed.
But as i store the angle/time tuples for my linear interpolation, i also smoothed the 

```
float HallSensor::getVelocity()
{
  long last_pulse_diff = pulse_diff;
  if (last_pulse_diff == 0 || ((long)(_micros() - pulse_timestamp) > 2*last_pulse_diff) ) // last velocity isn't accurate if too old
    return 0;

  float vel = direction * (_2PI / (float)cpr) / (last_pulse_diff / 1000000.0f);

  bNoUpdate = true;   // thread safety
  float fPulseDiff = last_pulse_diff;
  int iSum = 1;
  for(int i=1; i<PolynomialInterpolationCount-1; i++)
  {
    int iDiff = aiTimeLastUpdate[i] - aiTimeLastUpdate[i+1];
    if (  (fPulseDiff + iDiff) > 50000) // do not average over more than 50 ms
      break;
    fPulseDiff += iDiff; 
    iSum++;
  }
  if (bNoUpdate)
  {
    fPulseDiff /= iSum;
    vel = direction * (_2PI / (float)cpr) / (last_pulse_diff / 1000000.0f);
  }
    
  // quick fix https://github.com/simplefoc/Arduino-FOC/issues/192
  if(vel < -velocity_max || vel > velocity_max)  //if velocity is out of range then make it zero
    return 0;
  return vel;
}
```

I think that lowering the `driver.voltage_power_supply` will still allow that FOC black box to generate max sine curves.
Which afterwards get clamped to square waves in 

// Set voltage to the pwm pin
void BLDCDriver6PWM::setPwm(float Ua, float Ub, float Uc) {
  // limit the voltage in driver
  Ua = _constrain(Ua, 0, voltage_limit);
  Ub = _constrain(Ub, 0, voltage_limit);
  Uc = _constrain(Uc, 0, voltage_limit);
  // calculate duty cycle
  // limited in [0,1]
  dc_a = _constrain(Ua / voltage_power_supply, 0.0f , 1.0f );
  dc_b = _constrain(Ub / voltage_power_supply, 0.0f , 1.0f );
  dc_c = _constrain(Uc / voltage_power_supply, 0.0f , 1.0f );
  // hardware specific writing
  // hardware specific function - depending on driver and mcu
  _writeDutyCycle6PWM(dc_a, dc_b, dc_c, phase_state, params);
}

But that is just my guess.
That max speed now totally outperforms the Gen2 code.

RoboDurden, 2023-06-30:
The chatGPT code was a misfire. Did only produce overflow floats.
I found a nice Lagrange Interpolation the old google way.

```
  long iNow = _micros();
  float fAngleNow = 0;
  for(int i=0; i<PolynomialInterpolationCount; i++)
  {
    float fCoeff = 1;
    for(int j=0; j<PolynomialInterpolationCount; j++)
      if( i!=j )
        fCoeff *=  (float)(iNow - aiTimeLastUpdate[j]) / (float)(aiTimeLastUpdate[i] - aiTimeLastUpdate[j]);
    fAngleNow += fCoeff * aiAngleLastUpdate[i];
  }
  fAngleNow = 0.1 * fAngleNow + 0.9 * (float)(electric_rotations * 6 + electric_sector);    // only take 10% of the predictioin
  fAngleLagrange = (fmod(fAngleNow,cpr) / (float)cpr) * _2PI ;
  if (bNoUpdate)  
    return fAngleLagrange;
```

But adding mor then 10% of that 4th polynomial interpolation did block the motor :-(
(i logged the `fAngleLagrange` and `fAngleOrg` and could verify that the predicted `fAngleLagrange` was reasonable)

I also noticed that the execution time of that code affected the motor behavior (additional 30 mA) even without using `fAngleLagrange` at all. So the `getMechanicalAngle() ` call from `BLDCMotor::loopFOC()` is very time critical. 
And my simple linear interpolation might already be the optimum :-)

robcazzaro, 2023-06-30:
> I still have not undestood why setting `driver.voltage_power_supply` lower as the true battery voltage makes the motor even spin faster. And now with true voltage already exceeding Gen2, setting

The way I understand it, SimpleFOC needs to know the power supply voltage value when calculating currents (you can set the motor impedance) and to optimize voltage and current in each phase.

Then you use motor.voltage_limit and driver.voltage_limit to protect the motor especially when starting to protect the motor and board.

Let's say that you have a 10V power supply and set driver.voltage_power_supply to 10 and driver.voltage_limit to 5V. The PWM will never produce more than 50% voltage when the algorithm would output 100%. If you set driver.voltage_limit to 10, now at max phase voltage you will get 10V

It looks as if, by setting the power supply value lower than actual, you are driving the PWM more fully. Do you have motor.voltage_limit and driver.voltage_limit set, by chance? Try printing those values and see if they are the same or lower than the power supply value

Keep in mind that speed is only one aspect. For my uses, for example, I will need the motor to move very slowly (<10 RPM) with constant torque and be able to hold position with the same torque. The value of using SimpleFOC, for me, would be that once a board and motor parameters are optimized, it can be used in many different ways equally easy, without having to change the core code, just change settings in the setup()

Candas1, 2023-06-30:
I would check the 3 voltages with [this](https://community.simplefoc.com/t/using-stm-studio-for-variable-monitoring/3177?u=Candas1), maybe we see what happens

Candas1, 2023-06-30:
I am wondering if at high duty cycle the values are clamped, so the sinusoidal is flat.

robcazzaro, 2023-06-30:
> I am wondering if at high duty cycle the values are clamped, so the sinusoidal is flat.

The best way to verify that would be to use a logic analyzer connected to the processor PWM pins (it's enough to only use 3 pins, like the 3 high phase pins), and once the motor is spinning at constant speed, capture enough pulses to see the PWM signal. It's easy that way to see a clipped signal, as a long sequence of 1s

RoboDurden, 2023-07-01:
Thanks @robcazzaro , having two `voltage_limit `at the same time in different classes looks like very bad programming behavior to me. 
`FOCMotor::voltage_limit` is never used in that base class `FOCMotor` class anyway and the dereived `BLDCMotor` can not work without a driver and also makes multiple uses of `driver->voltage_limit`.
So i guess it is a bug to use `FOCMotor::voltage_limit` in some other places of the same code file :-(
If indeed some other derived `FOCMotor` class would need its own `voltage_limit` without having a driver, then it should be defined in that other class and not in the base class.

```
OUT2T( motor.voltage_limit, driver.voltage_limit)
-> 12.00: 25.20
```

The 25.2 Volt is my 7s 3.6 * BAT_CELLS, the 12.0 is still the DEF_POWER_SUPPLY

Indeed, in `BLDCMotor::move(float new_target)`, the `motor::voltage_limit` is used for clamping:
```
  switch (controller) {
    case MotionControlType::torque:
      if(torque_controller == TorqueControlType::voltage){ // if voltage torque control
        if(!_isset(phase_resistance))  voltage.q = target;
        else  voltage.q =  target*phase_resistance + voltage_bemf;
        voltage.q = _constrain(voltage.q, -voltage_limit, voltage_limit);
```
But `BLDCMotor::setPhaseVoltage` concludes with   `driver->setPwm(Ua, Ub, Uc);` and there the 
`Ua = _constrain(Ua, 0, voltage_limit);`
has no effect because `driver::voltage_limit` is way higher than the still default `12.0` used for clamping in `BLDCMotor::move()`.
And the final clamp to `[0.0..1.0]` i guess will only allow the max of `1.0` if i also reduce the `driver::voltage_power_supply` to 12.0 V
`  dc_a = _constrain(Ua / voltage_power_supply, 0.0f , 1.0f );`

I think there should not be a `FOCMotor::voltage_limit` at all :-(
It is never used in that base class `FOCMotor`, and in the derrived `BLDCMotor` class `driver->setPwm(Ua, Ub, Uc);` is hard coded in `BLDCMotor::setPhaseVoltage(...)`. So it would be no problem at all to not use `FOCMotor::voltage_limit` in `BLDCMotor::move()` (and other places) but 
`        voltage.q = _constrain(voltage.q, -driver->voltage_limit, driver->voltage_limit);`
instead of what is now implemented:
`        voltage.q = _constrain(voltage.q, -voltage_limit, voltage_limit);`

I also do not (want) to undestand why using voltages when in the end we factor and clamp it down to [0..1] anyway.
The `voltage_power_supply` is only used to scale the duty cacle down to 0..1. There is no analog read of the battery voltage to compare it against `voltage_power_supply`. in neither the base class `BLDCDriver` nor any of the derived `BLDCDriver_x_PWM classes`.
The [0.0..100.0] duty cycle is more intuitive.

But i think it is too late to change that now.
At least the `FOCMotor:.voltage_limit` should be removed i guess.

Do not really have the happiness to start arguing with the simpleFOC community..
@Candas1 , why don't you accept my pull request: https://github.com/Candas1/Split_Hoverboard_SimpleFOC/pull/7  ?
Then @robcazzaro could already begin with adc low level stuff.
I do not really have a use scenario that needs currents sensing. I feel current sensing is NOT needed for FOC but only for torque control..

Candas1, 2023-07-01:
Because I don't aim to maintain my own branch of simplefoc except the GD32 drivers. 
Anything else will also be beneficial to the simpleFOC community. 
You are working on topics the simpleFOC team is already about to release, without even trying it. 
This will even make it harder to merge anything new they are bringing.

RoboDurden, 2023-07-01:
I don't pose a problem if I can't offer any solutions
* i am totally undecided whether FOCMotor::voltage_limit or BLDCDriver_x_PWM::voltage_limit should be removed.
* I don't know if removing that unneccesary voltage stuff could be removed anyway
* I only have a stupid linear interpolation that boosts max speed by 33% but performs poorly at low speeds.

So no solutions to offer and therefore no happiness to pose questions to other people.

RoboDurden, 2023-07-01:
@robcazzaro i still do not really understand why you don't buy some used hoverboards locally. I do not know all the US websites to buy used items but already facebooks lists multiple items: https://www.facebook.com/marketplace/seattle/hoverboards?maxPrice=30&exact=false
And me (and i guess Candas too) would be happy to each $20 donate  to you for getting some boards.
If your really plan to do multiple projects with hoverbaords you will need your own stock of used hoverboards anyway.

RoboDurden, 2023-07-01:
i am not really making progress with my linear interpolation. Has taken all day to verify that my angle prediction works quite good, but the motor does not like the "better" values.
I also notice that the motor is spinning more poorly left then right.
Sometimes it even gets "out of phase" and is rotating with a bad stutter. Then after a few seconds it kicks back "into the right phase" and rotates "nicely" again.
So the original HallSensor is also not perfect.

With Gen2 however, the phase pwm are simply directly calculated from the 6 (8?) possible hall combination. 
```
  // Determine current position based on hall sensors
  hall = hall_a * 1 + hall_b * 2 + hall_c * 4;
  pos = hall_to_pos[hall];
```
So nothing can get "out of phase" over time.

With simpleFOC however, the angle increments with every hall step, is then mapped to 360° = [0.0 .. 2pi], some offset is added
`sensor->getMechanicalAngle()  - zero_electric_angle` 
but once that angle "gets out of phase", the three pwm phase voltages 
get applied wrong "forever".

I think there should be an ongoing closed loop that corrects the `zero_electric_angle`.
to match the coarse `hall_to_pos[hall];`
And i guess there should be a `zero_electric_angle_left` and `zero_electric_angle_right`

@Candas1 , how did you obtain `motor.initFOC(2.09,Direction::CCW); // Start FOC without alignment` which skips the `BLDCMotor::alignSensor()`- which does not work at all for me. Only your `2.09` work
Was that simply try and error ? Then maybe with my "better" linear interpolation, i must find a new value.

Candas1, 2023-07-01:
Please check those comments:
https://github.com/Candas1/Split_Hoverboard_SimpleFOC/issues/1#issuecomment-1592731286
https://github.com/Candas1/Split_Hoverboard_SimpleFOC/issues/5#issuecomment-1605016055
https://github.com/Candas1/Split_Hoverboard_SimpleFOC/issues/5#issuecomment-1610263400

robcazzaro, 2023-07-01:
@RoboDurden SimpleFOC is a bit hard to understand and has quite a few quirks. On the other hand, it's solid code that works for a variety of projects and hardware. Trying to hack around its inner workings without understanding why certain things were done a certain way it's a recipe for frustration. Many of the issues we are facing are known by the SimpleFOC team and work is happening to address them.

If we think something could be done better, the right approach is to engage with them and suggest changes. Most of the times, their answer will explain why their approach is better and the side effects of doing things differently. Every time I suggested changes, they were incredibly open and responsive.

You say `I do not really have a use scenario that needs currents sensing. I feel current sensing is NOT needed for FOC but only for torque control..` Well, torque control at very low speeds is the only scenario that matters to me 😁, and the reason why we even started having these conversations. I'd like to have a solution for torque control using the hoverboard split boards. I already have other boards (like DRV8302 that works perfectly well with SimpleFOC, just wanted to see if it's possible to use a split hoverboard board

So, while I appreciate your desire to quickly get the code to a working state for higher speeds, that is not an approach I think will work long term. As I mentioned before, there is already a solution for a smoothed hall sensor that works at any speed. It might be suboptimal in certain cases compared to your approach, but works in general. IF you think your approach is better, I'm sure that the SimpleFOC people will be interested in hearing about it and discuss it with you.

[14 more replies]