Candas1/Split_Hoverboard_SimpleFOC #6: adding LowsideCurrentSense ...

RoboDurden, 2023-06-29:
In continuation of https://github.com/Candas1/Split_Hoverboard_SimpleFOC/issues/5

I have added the needed code to my cloned fork but can not push because Github_Desktop hangs and therefore can not make a pull request.
But we might already talk about the best approach to add LowsideCurrentSense:

```
...
// BLDC motor & driver instance
BLDCMotor motor = BLDCMotor(BLDC_POLE_PAIRS);
BLDCDriver6PWM driver = BLDCDriver6PWM( BLDC_BH_PIN,BLDC_BL_PIN,  BLDC_GH_PIN,BLDC_GL_PIN,  BLDC_YH_PIN,BLDC_YL_PIN );

// shunt resistor value , gain value,  pins phase A,B,C
LowsideCurrentSense current_sense = LowsideCurrentSense(BLDC_CUR_Rds, BLDC_CUR_Gain, BLDC_CUR_G_PIN, BLDC_CUR_B_PIN, BLDC_CUR_Y_PIN);
```

```
setup()
{
...
/*  
  // succeeds but motor.initFOC(2.09,Direction::CCW)  hangs 
  // motor.initFOC() will turn motor forward and backward a bit and then also hangs
  // We probably need a SimpleFOC/src/current_sense/hardware_specific/gd32/gd32_mcu.cpp
  if (current_sense.init())
  {
    motor.linkCurrentSense(&current_sense);
    Blink(3,oLedOrange);
  }
  else
  {
    Blink(5);
    OUTN("current_sense.init() failed.")
    return;   // cancel simpleFOC setup
  }
*/
```
```

loop()
{
...
    if (current_sense.initialized)
    {
      PhaseCurrent_s currents = current_sense.getPhaseCurrents();
      float current_magnitude = current_sense.getDCCurrent();
      OUT2T("mA",current_magnitude*1000)  // milli Amps
      OUT2T("B mA",currents.b*1000)  // milli Amps
      OUT2T("C mA",currents.c*1000)  // milli Amps
    }

```

Candas1, 2023-07-11:
There is still a lot for work to be done on the 6PWM driver before starting to work on current sensing.

RoboDurden, 2023-07-11:
You mean that the init code of the 6 pwm drivers need to be improved ? That of course would impact my speed tests :-)
Or do you want to complete the gd32_mcu.cpp to support different timers (some different boards) or even add other motors like steppers ?
I see these functions in stm32_mcu.cpp that i don't find in your (and @robcazzaro  ?) nice gd32_mcu.cpp:

```
_configure1PWM
_configure2PWM
_configure3PWM
_configure4PWM
_writeDutyCycle2PWM
_writeDutyCycle3PWM
_writeDutyCycle4PWM

_initPinPWM
_initPinPWMHigh
_initPinPWMLow

_stopTimers
_startTimers

void _alignPWMTimers(HardwareTimer *HT1, HardwareTimer *HT2, HardwareTimer *HT3);
void _alignPWMTimers(HardwareTimer *HT1, HardwareTimer *HT2, HardwareTimer *HT3, HardwareTimer *HT4);
_alignTimersNew

_initHardware6PWMPair
_initHardware6PWMInterface

int findBestTimerCombination(int numPins, int index, int pins[], PinMap* pinTimers[]);
int findBestTimerCombination(int numPins, int pins[], PinMap* pinTimers[]);
scoreCombination
_isChannelEnabled
findIndexOfFirstPinMapEntry
findIndexOfLastPinMapEntry

printTimerCombination
getTimerNumber
```

finding available/optimal timers might be good for supporting multiple bldc motors like with Gen1 boards.

robcazzaro, 2023-07-12:
I think that @Candas1 meant that the GD32 implementation of 6PWM is not complete. For example, we don't correctly process the various flags for the signal polarity. It's an item on my todo list. But we won't improve performance by doing so. After all, the 6PWM is simply setting up a timer for the complementary PWM signals of the 3 phases. Once the timer is properly set, there're no further gains to have: a timer is a timer, and PWM is PWM. It's the rest of the code that can be improved.

Gen1 boards do not use the GD32F130, as far as I know, so there's no value in doing the other PWMs. The Gen 2 boards only support 6PWM. I mean, it might be good to have a truly generic GD32 implementation, but before we spend the time, we might as well ensure that the Gen2 boards work properly. Once that works, if there is interest from the community, we could consider adding the remaining PWM types.

STM32 implements all those types because there are widely used STM32 boards (e.g. BluePill, Nucleo, etc) that can be connected to any type of drivers. There are no GD32F130 boards available, and very few GD32F103 ones as well

RoboDurden, 2023-07-19:
So i have begun to port the Gen2 adc code and maybe @robcazzaro would like to help me with my low-level code problems. 
@Candas1 only read this if you intentionally droped the update event handler init code while porting from Gen2 to your nice drivers\hardware_specific\gd32\gd32_mcu.cpp.

I now think that adc sampling is not triggered by a new timer but by the bldc pwm timer0 !

```
//----------------------------------------------------------------------------
// Timer0_Update_Handler
// Is called when upcouting of timer0 is finished and the UPDATE-flag is set
// AND when downcouting of timer0 is finished and the UPDATE-flag is set
// -> pwm of timer0 running with 16kHz -> interrupt every 31,25us
//----------------------------------------------------------------------------
void TIMER0_BRK_UP_TRG_COM_IRQHandler(void)
{
    LowsideTimerCurrentSense::iCount1++;

	// Start ADC conversion
	adc_software_trigger_enable(ADC_REGULAR_CHANNEL);
	
	// Clear timer update interrupt flag
	timer_interrupt_flag_clear(TIMER_BLDC, TIMER_INT_UP);
}
```

Yes i began with a new sub class `LowsideTimerCurrentSense` and a static member `iCount1` to check if this functions get called.

The Gen2 init code for adc compiles nicely and if the TIMER_BLDC triggered TIMER0_BRK_UP_TRG_COM_IRQHandler this handler function then calls

```
//----------------------------------------------------------------------------
// This function handles DMA_Channel0_IRQHandler interrupt
// Is called, when the ADC scan sequence is finished
// -> ADC is triggered from timer0-update-interrupt -> every 31,25us
//----------------------------------------------------------------------------
void DMA_Channel0_IRQHandler(void)
{
    LowsideTimerCurrentSense::iCount2++;
	// Calculate motor PWMs
	//CalculateBLDC();
	
	if (dma_interrupt_flag_get(DMA_CH0, DMA_INT_FLAG_FTF))
	{
		dma_interrupt_flag_clear(DMA_CH0, DMA_INT_FLAG_FTF);        
	}
}
```

I monitor the two `iCountX` in main.c but they stay at 0.

`    OUT2T(current_sense.iCount1, current_sense.iCount2 )
`

Now i do not see crucial changes in the pwm port from Candas to GD32F130C8\Simple FOC\src\drivers\hardware_specific\gd32\gd32_mcu.cpp

Only thing that was missing (droped by Candas) was

```
	nvic_irq_enable(TIMER0_BRK_UP_TRG_COM_IRQn, 0, 0);
	timer_interrupt_enable(TIMER_BLDC, TIMER_INT_UP);
```

But `TIMER0_BRK_UP_TRG_COM_IRQHandler()` does not get called :-/

When i place that function in two different files i get a linker error of mulitple definitions. So my one function does get compiled and linked into the binary.

Now TIMER0_BRK_UP_TRG_COM_IRQHandler might be a Keil specific naming convention defined in Gen2 startup_gd32f1x0.s

```
                DCD     ADC_CMP_IRQHandler                ; 28:ADC and Comparator 0-1
                DCD     TIMER0_BRK_UP_TRG_COM_IRQHandler  ; 29:TIMER0 Break,Update,Trigger and Commutation
[... 198 lines cut ...]
    .channels = { TIMER_BLDC_CHANNEL_G, TIMER_BLDC_CHANNEL_G, TIMER_BLDC_CHANNEL_B, TIMER_BLDC_CHANNEL_B, TIMER_BLDC_CHANNEL_Y, TIMER_BLDC_CHANNEL_Y },
    .pwm_frequency  = pwm_frequency,
    .range          = SystemCoreClock/pwm_frequency,
    .dead_zone      = dead_zone,
    .interface_type = _HARDWARE_6PWM
  };
```

And my latest handler function:

```
void TIMER0_IRQHandler(void)
{
	LowsideTimerCurrentSense::iCount1++;
	//iCounter3++;

	// clear interrupt request to enable next run
	if (timer_interrupt_flag_get(TIMER0, TIMER_INT_FLAG_UP) != RESET) 
	{
		timer_interrupt_flag_clear(TIMER0, TIMER_INT_FLAG_UP);
	}
}

```
Ideas welcome :-)

robcazzaro, 2023-07-19:
@RoboDurden yes, happy to help. Please let me know exactly what you'd like me to look into.

But, please, keep in mind that the right way to handle low side current sampling is by using the inserted ADC functionality (injected for STM32), not using a very slow software interrupt timer. That's what the STM32 driver does. We are working with a 48MHz processor, and should avoid doing in software what can be done in hardware

RoboDurden, 2023-07-19:
The Timer2 example from the GD32-Arduino community works. And i guess it is better anyway to keep driver code seperated from current_sensor code.

```
#include <Arduino.h>

#define PIN_LED PB3

uint32_t getTimerClkFrequency(uint32_t instance)
{
    rcu_clock_freq_enum timerclkSrc = 0;
    uint32_t APBx_PSC = 0;
    uint32_t clk_src = 0;
    if (instance != (uint32_t) 0) {
        switch ((uint32_t)instance) {
            case (uint32_t)TIMER0:
            case (uint32_t)TIMER1:
            case (uint32_t)TIMER2:
            case (uint32_t)TIMER5:
            case (uint32_t)TIMER13:
                timerclkSrc = CK_APB1;
                APBx_PSC = (RCU_CFG0 & RCU_CFG0_APB1PSC) >> 8;
                break;
            default:
                break;
        }
    }
    if (0 != (APBx_PSC & 0x04)) {
        clk_src = 2 * rcu_clock_freq_get(timerclkSrc);
    } else {
        clk_src =  rcu_clock_freq_get(timerclkSrc);
    }
    return clk_src;
}

void config_timer(uint32_t timer_periph) {
    nvic_irq_enable(TIMER2_IRQn, 2, 2); // hardcode enable Timer2 interrupt in NVIC
    /* enable clock input for Timer2 peirpheral */
    rcu_periph_clock_enable(RCU_TIMER2);
    /* configure TIMER base function */
    timer_parameter_struct timer_initpara;
    uint16_t every_x_milliseconds = 200;
    timer_initpara.prescaler = getTimerClkFrequency(timer_periph) / 10000 - 1;
    timer_initpara.period = every_x_milliseconds * 10 - 1;
    timer_initpara.repetitioncounter = 0;
    timer_initpara.clockdivision = TIMER_CKDIV_DIV1;
    timer_initpara.counterdirection = TIMER_COUNTER_UP;
    timer_initpara.alignedmode = TIMER_COUNTER_EDGE;
    timer_autoreload_value_config(timer_periph, 0);
    timer_init(timer_periph, &timer_initpara);
    /* enable timer update interrupt */ 
    /* happens when timer is overflowing (period / reload value exceeded) and it is set back to 0 again */
    timer_interrupt_flag_clear(timer_periph, TIMER_INT_FLAG_UP);
    timer_interrupt_enable(timer_periph, TIMER_INT_UP);
    timer_enable(timer_periph);
}

int main(void)
{
    pinMode(PIN_LED, OUTPUT);
    config_timer((uint32_t)TIMER2);
    while (1)
    {
        __WFI(); // wait for interrupt. not needed but debugging is easier :)
    }
}

uint8_t state = 0;
void TIMER2_IRQHandler(void)
{
    // clear interrupt request to enable next run
    if (timer_interrupt_flag_get(TIMER2, TIMER_INT_FLAG_UP) != RESET) {
        timer_interrupt_flag_clear(TIMER2, TIMER_INT_FLAG_UP);
        /* our own reaction to timer update interrupt triggering: invert GPIO pin, blinky blink */
        digitalWrite(PIN_LED, state);
        state ^= 1; // invert for next run
    }
}

void NMI_Handler(void) {}
void HardFault_Handler(void){   while (1);  }
void MemManage_Handler(void){   while (1);  }
void BusFault_Handler(void){   while (1);  }
void UsageFault_Handler(void){   while (1);  }
void SVC_Handler(void){}
void DebugMon_Handler(void){}
void PendSV_Handler(void){}
//void SysTick_Handler(void){}
```

>not using a very slow software interrupt timer. 

Don't understand why you two keep on repeating this.
The "Gen2 style" is to only trigger the adc from a timer interrupt. And if

```
// -> pwm of timer0 running with 16kHz -> interrupt every 31,25us
//----------------------------------------------------------------------------
void TIMER0_BRK_UP_TRG_COM_IRQHandler(void)
```

is correct then 

```
// This function handles DMA_Channel0_IRQHandler interrupt
// Is called, when the ADC scan sequence is finished
// -> ADC is triggered from timer0-update-interrupt -> every 31,25us
//----------------------------------------------------------------------------
void DMA_Channel0_IRQHandler(void)

```
should also succeed with 2 adc (battery voltage and overall current) at a 16 kHz rate.

And when adc is complete it is also the best time to call `motor.loopFOC();` :-)

Calling this function from the main loop only achieved about 5kHz with exceptions of below 1 kHz :-(
So i do not need the 16 kHz of the pwm timer0 (if that 16 kHz is correct).
With a seperate Timer2 i can trigger the adc at lets say 8 kHz and increased the adc from 2 to 4 (low_side current) :-)

Will continue tomorrow.

RoboDurden, 2023-07-20:
Could no longer flash my board so i had to solder a little wire from the NRST mcu pin to the RST header pin. inside one of my ST-Link dongles: https://modwiggler.com/forum/viewtopic.php?p=1999431#p1999431
The RST pin of these cheap clones only connects to the SWIM flash pins used for the STM8 MCUs. I also did unsolder a little smd resistor on the backside to use the RST output pin exclusively for the SWD protocol.

When i copy the working TIMER2 code (see previous post) into my Split_Hoverboard_SimpleFOC project, it does no longer work.
When i completely replace the main.cpp and the platformio.ini::[env] i get a compiler error

```
Compiling .pio\build\xxGD32F130C8\src\main.cpp.o
src\main.cpp: In function 'uint32_t getTimerClkFrequency(uint32_t)':
src\main.cpp:7:39: error: invalid conversion from 'int' to 'rcu_clock_freq_enum' [-fpermissive]
    7 |     rcu_clock_freq_enum timerclkSrc = 0;
```

Is there some other ini file in our Split_Hoverboard_SimpleFOC  project that sets some compiler directives ?

```
[env:xxGD32F130C8]
debug_tool = stlink
upload_protocol = stlink
framework = arduino
platform = https://github.com/CommunityGD32Cores/platform-gd32.git
platform_packages = framework-arduinogd32@https://github.com/CommunityGD32Cores/ArduinoCore-GD32.git
board = genericGD32F130C8
build_flags = -D __PIO_DONT_SET_CLOCK_SOURCE__
	-D __SYSTEM_CLOCK_48M_PLL_IRC8M_DIV2=48000000
	-D $PIOENV 	
lib_deps = 
lib_archive = false
monitor_port = socket://localhost:9090
monitor_filters = send_on_enter
monitor_eol = LF
monitor_echo = yes
```

Ideas welcome.

RoboDurden, 2023-07-20:
Okay :-(
The working GD32-Arduino Timer2 example https://github.com/CommunityGD32Cores/gd32-pio-projects/tree/main/gd32-spl-timer has a main.c and not a main.cpp. Therefore it gets compile a C-code.
And i can not add simpleFOC to a main.c project because that is a nice c++ library.

But the Timer example no longer works when compiled as c++ (rename to main.cpp):

```
void config_timer(uint32_t timer_periph) {
    nvic_irq_enable(TIMER2_IRQn, 2, 2); // hardcode enable Timer2 interrupt in NVIC
    // enable clock input for Timer2 peirpheral
    //rcu_periph_clock_enable(RCU_AF);
    rcu_periph_clock_enable(RCU_TIMER2);
    // configure TIMER base function
    timer_parameter_struct timer_initpara;
    uint16_t every_x_milliseconds = 200;
    timer_initpara.prescaler = getTimerClkFrequency(timer_periph) / 10000 - 1;
    timer_initpara.period = every_x_milliseconds * 10 - 1;
    timer_initpara.repetitioncounter = 0;
    timer_initpara.clockdivision = TIMER_CKDIV_DIV1;
    timer_initpara.counterdirection = TIMER_COUNTER_UP;
    timer_initpara.alignedmode = TIMER_COUNTER_EDGE;
    timer_autoreload_value_config(timer_periph, 0);
    timer_init(timer_periph, &timer_initpara);
    // enable timer update interrupt
    // happens when timer is overflowing (period / reload value exceeded) and it is set back to 0 again
    timer_interrupt_flag_clear(timer_periph, TIMER_INT_FLAG_UP);
    timer_interrupt_enable(timer_periph, TIMER_INT_UP);
    timer_enable(timer_periph);
}
...
uint8_t state = 0;
void TIMER2_IRQHandler(void)
{
    // clear interrupt request to enable next run
    if (timer_interrupt_flag_get(TIMER2, TIMER_INT_FLAG_UP) != RESET) {
        timer_interrupt_flag_clear(TIMER2, TIMER_INT_FLAG_UP);
        // our own reaction to timer update interrupt triggering: invert GPIO pin, blinky blink
        digitalWrite(PIN_LED, state);
        state ^= 1; // invert for next run
    }
}
```

Then the main.cpp code compiles but the led is not blinking. And F5 debug confirms that the interrupt handler never gets called.

@robcazzaro , you might have an idea how to port this c-code example to c++ :-)

My platformio.ini for both, the main.c and main.cpp project:

```
[env:GD32F130C8]
debug_tool = stlink
upload_protocol = stlink
framework = arduino
platform = https://github.com/CommunityGD32Cores/platform-gd32.git
platform_packages = framework-arduinogd32@https://github.com/CommunityGD32Cores/ArduinoCore-GD32.git
board = genericGD32F130C8
build_flags = -D __PIO_DONT_SET_CLOCK_SOURCE__
	-D __SYSTEM_CLOCK_48M_PLL_IRC8M_DIV2=48000000
	-D $PIOENV 	
lib_deps = 
lib_archive = false
monitor_port = socket://localhost:9090
monitor_filters = send_on_enter
monitor_eol = LF
monitor_echo = yes
```

RoboDurden, 2023-07-20:
For sure i strongly dislike this low-level c-style code :-(
After searching the net for interrupt handlers in c++ and Arduino or PlatformIO i finally learned:

there is already a nice `#include <HardwareTimer.h>` that is supported by GD32-Arduino

```
#include <Arduino.h>
#include <HardwareTimer.h>
#include <SimpleFOC.h>

#define LED PB3
#define LED_GREEN PA15

HardwareTimer t(TIMER2);

unsigned long iCount = 0;
#define WINDOW 16000

void timer_cb() 
{
    digitalWrite(LED, digitalRead(LED) ^ 1);
	iCount++;
    digitalWrite(LED_GREEN, (iCount % WINDOW) < (WINDOW/2) );
}

void setup()
{
    pinMode(LED, OUTPUT);
    pinMode(LED_GREEN, OUTPUT);
    t.setPeriodTime(16000, FORMAT_HZ);
    t.attachInterrupt(&timer_cb);
    t.start();
}

void loop() {}
```

Of course now i fear that simply adding 

```
	// Start ADC conversion
	adc_software_trigger_enable(ADC_REGULAR_CHANNEL);
```

to that `void timer_cb()` will not call 

```
//----------------------------------------------------------------------------
// This function handles DMA_Channel0_IRQHandler interrupt
// Is called, when the ADC scan sequence is finished
// -> ADC is triggered from timer0-update-interrupt -> every 31,25us
//----------------------------------------------------------------------------
void DMA_Channel0_IRQHandler(void)
{
	motor.loopFOC();
	
	if (dma_interrupt_flag_get(DMA_CH0, DMA_INT_FLAG_FTF))
	{
		dma_interrupt_flag_clear(DMA_CH0, DMA_INT_FLAG_FTF);        
	}
}
```
 
because that probably is also stupid c-style code :-(

RoboDurden, 2023-07-20:
Yes, i will port that Gen2 code https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/HoverBoardGigaDevice/Src/setup.c#L336
And yes, the DMA IRQ will/should be called at the end of the conversion.
But in Gen2 the conversion begin is triggered from the TIMER0 that is primarily used for the bldc pwm.
As i could not get the Candas-port to simpleFOC make the TIMER0 call an event handler like Gen2 to trigger/begin the conversion https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/HoverBoardGigaDevice/Src/it.c#L126

I will now use a seperate TIMER2 with that nice GD32-Arduino. Does not really matter from where the adc conversion is triggered.

But i fear that the DMA IRQ will not get call in c++ just like the TIMER0 handler did not get called.

RoboDurden, 2023-07-20:
of course the DMA IRQ handler gets not called:

```
#include <Arduino.h>
#include <HardwareTimer.h>
#include <SimpleFOC.h>

#define LED_GREEN PA15
#define LED_ORANGE PA12
#define LED_RED PB3

HardwareTimer t(TIMER2);

unsigned long iCount = 0;
#define TIMER2_HZ 8000

void timer_cb() 
{
    digitalWrite(LED_RED, digitalRead(LED_RED) ^ 1);
	iCount++;
    digitalWrite(LED_ORANGE, (iCount % TIMER2_HZ) < (TIMER2_HZ/4) );

	// Start ADC conversion
	adc_software_trigger_enable(ADC_REGULAR_CHANNEL);
}

// This function handles DMA_Channel0_IRQHandler interrupt
// Is called, when the ADC scan sequence is finished
void DMA_Channel0_IRQHandler(void)
{
	//motor.loopFOC();
    digitalWrite(LED_GREEN, (iCount % TIMER2_HZ) < (TIMER2_HZ/2) );

	if (dma_interrupt_flag_get(DMA_CH0, DMA_INT_FLAG_FTF))
		dma_interrupt_flag_clear(DMA_CH0, DMA_INT_FLAG_FTF);        
}

// ADC defines
#define VBATT_PIN	GPIO_PIN_4
#define VBATT_PORT GPIOA
#define VBATT_CHANNEL ADC_CHANNEL_4
#define CURRENT_DC_PIN	GPIO_PIN_6
#define CURRENT_DC_PORT GPIOA
#define CURRENT_DC_CHANNEL ADC_CHANNEL_6

// DMA (ADC) structs
dma_parameter_struct dma_init_struct_adc;
// ADC buffer struct
typedef struct
{
  uint16_t v_batt;
	uint16_t current_dc;
} adc_buf_t;

adc_buf_t adc_buffer;

//----------------------------------------------------------------------------
// Initializes the ADC
//----------------------------------------------------------------------------
void ADC_init(void)
{
	gpio_mode_set(VBATT_PORT, GPIO_MODE_ANALOG, GPIO_PUPD_NONE, VBATT_PIN);
	gpio_mode_set(CURRENT_DC_PORT, GPIO_MODE_ANALOG, GPIO_PUPD_NONE, CURRENT_DC_PIN);

	// Enable ADC and DMA clock
	rcu_periph_clock_enable(RCU_ADC);
	rcu_periph_clock_enable(RCU_DMA);
	
  // Configure ADC clock (APB2 clock is DIV1 -> 72MHz, ADC clock is DIV6 -> 12MHz)
	rcu_adc_clock_config(RCU_ADCCK_APB2_DIV6);
	
	// Interrupt channel 0 enable
	nvic_irq_enable(DMA_Channel0_IRQn, 1, 0);
	
	// Initialize DMA channel 0 for ADC
	dma_deinit(DMA_CH0);
	dma_init_struct_adc.direction = DMA_PERIPHERAL_TO_MEMORY;
	dma_init_struct_adc.memory_addr = (uint32_t)&adc_buffer;
	dma_init_struct_adc.memory_inc = DMA_MEMORY_INCREASE_ENABLE;
	dma_init_struct_adc.memory_width = DMA_MEMORY_WIDTH_16BIT;
	dma_init_struct_adc.number = 2;
	dma_init_struct_adc.periph_addr = (uint32_t)&ADC_RDATA;
	dma_init_struct_adc.periph_inc = DMA_PERIPH_INCREASE_DISABLE;
	dma_init_struct_adc.periph_width = DMA_PERIPHERAL_WIDTH_16BIT;
	dma_init_struct_adc.priority = DMA_PRIORITY_ULTRA_HIGH;
	dma_init(DMA_CH0, &dma_init_struct_adc);
	
	// Configure DMA mode
	dma_circulation_enable(DMA_CH0);
	dma_memory_to_memory_disable(DMA_CH0);
	
	// Enable DMA transfer complete interrupt
	dma_interrupt_enable(DMA_CH0, DMA_CHXCTL_FTFIE);
	
	// At least clear number of remaining data to be transferred by the DMA 
	dma_transfer_number_config(DMA_CH0, 2);
	
	// Enable DMA channel 0
	dma_channel_enable(DMA_CH0);
	
	adc_channel_length_config(ADC_REGULAR_CHANNEL, 2);
	adc_regular_channel_config(0, VBATT_CHANNEL, ADC_SAMPLETIME_13POINT5);
	adc_regular_channel_config(1, CURRENT_DC_CHANNEL, ADC_SAMPLETIME_13POINT5);
	adc_data_alignment_config(ADC_DATAALIGN_RIGHT);
	
	// Set trigger of ADC
	adc_external_trigger_config(ADC_REGULAR_CHANNEL, ENABLE);
	adc_external_trigger_source_config(ADC_REGULAR_CHANNEL, ADC_EXTTRIG_REGULAR_NONE);
	
	// Disable the temperature sensor, Vrefint and vbat channel
	adc_tempsensor_vrefint_disable();
	adc_vbat_disable();
	
	// ADC analog watchdog disable
	adc_watchdog_disable();
	
	// Enable ADC (must be before calibration)
	adc_enable();
	
	// Calibrate ADC values
	adc_calibration_enable();
	
	// Enable DMA request
	adc_dma_mode_enable();
    
	// Set ADC to scan mode
	adc_special_function_config(ADC_SCAN_MODE, ENABLE);
}

void setup()
{
    t.setPeriodTime(TIMER2_HZ, FORMAT_HZ);
    t.attachInterrupt(&timer_cb);
    t.start();
    pinMode(LED_RED, OUTPUT);
    pinMode(LED_GREEN, OUTPUT);
    pinMode(LED_ORANGE, OUTPUT);
}

void loop() {}

```
ideas welcome :-/

P.S. there is an GD32-Arduino example that one by one triggers adc conversion and then stupidly wait for the conversion to end. Maybe it would be sufficient to trigger all wanted adc conversion and simply check with the TIMER2-callback when all conversions have finished:

https://github.com/CommunityGD32Cores/gd32-pio-projects/blob/main/gd32-spl-adc-polling-gd32f1x0/src/main.c#L162C21-L162C21

robcazzaro, 2023-07-20:
@RoboDurden if I understand what you are doing, you are mixing Arduino code and low level SPL code from the old project. That might result in either the ADC or the DMA not being properly initialized, so either the EDC conversion doesn't happen or the DMA IRQ is not properly set to happen.

You will need to single step thru the code to see if the ADC, DMA and IRQ registers are set the right way. Or at least start by manually calling the ADC conversion and see if it works and transfers the data into the DMA structures. My guess is that you will see that the ADC conversion is not happening, hence no DMA IRQ is generated. If the ADC/DMA works when called from your code, check if the timer is firing as expected

RoboDurden, 2023-07-20:
Ah thanks @robcazzaro that's nice, i can already monitor the `adc_buffer` at `dma_init_struct_adc.memory_addr` to see if the trigger `adc_software_trigger_enable(ADC_REGULAR_CHANNEL);` in my TIMER2 callback does indeed kickoff the adc conversion.
When that succeeds i can start worrying about why the adc-conversion-finished handler does not get called.
But maybe the conversion is always finished by the time the TIMER2 callback gets fired again and than i can call `motor.loopFOC()` from that TIMER2 handler :-)
But depending on how fast the adc will be, the values might that way already be up to 200 microseconds old (TIMER2 with 5 kHz)
Will adc_buffer always hold the old values until they are replaced with new ones ? With a 32bit mcu i guess the uint16 will be written in one step. But i see in the Gen2 code that the trigger of a new conversion takes place after `adc_buffer.adc_buffer.current_dc` etc is read from the bldc updater:

```
void DMA_Channel0_IRQHandler(void)
{
	CalculateBLDC();
	
	if (dma_interrupt_flag_get(DMA_CH0, DMA_INT_FLAG_FTF))
		dma_interrupt_flag_clear(DMA_CH0, DMA_INT_FLAG_FTF);        
}
```

But i guess i will see when i monitor `adc_buffer.v_batt` and the values go crazy :-)

RoboDurden, 2023-07-20:
No, the adc_buffer values do not change:

```
void setup()
{
	#ifdef DEBUG_STLINK
	SimpleFOCDebug::enable(&rtt);
	#endif
	OUTN("Hello gd32-adc :-)")

    t.setPeriodTime(TIMER2_HZ, FORMAT_HZ);
    t.attachInterrupt(&timer_cb);
    t.start();
    pinMode(LED_RED, OUTPUT);
    pinMode(LED_GREEN, OUTPUT);
    pinMode(LED_ORANGE, OUTPUT);

	adc_buffer.v_batt = adc_buffer.current_dc = 42;
}

void loop() 
{
	OUT2T(adc_buffer.v_batt,adc_buffer.current_dc);
	OUTN(iCount);
	delay(10);
}
```

```
42: 42  147663
42: 42  147743
42: 42  147823
42: 42  147903
42: 42  147983
42: 42  148063
42: 42  148143
42: 42  148223
```

I guess i will make a break for a few days. Beginning a new journey with my 10 km/h solar camper on saturday.
[image]

But in two days i will make a stop at my train station for a few days and will continue there.

robcazzaro, 2023-07-20:
Yes, that's one of the concerns with a slow processor and all the Arduino framework overhead. 

Don't forget that Gen2 on the GD32 was overclocked to 72MHz, and we are running at 48MHz. So you need to make sure that the timer initialization code takes that into account.

If the ADC transfers values via DMA to adc_buffer, your code will always find valid values in that variable. Worst case, the battery ADC and current adc will be from different read cycles, but it makes no real difference since those are not related at all. As a matter of fact, the code is wasting a lot of time converting the battery voltage so frequently, I would suggest checking the battery no more than a few times per second, if that. 

Given your last message (no valid ADC values), it looks as if the initialization didn't work as expected

robcazzaro, 2023-07-20:
Enjoy your trip @RoboDurden 🌞

RoboDurden, 2023-07-20:
@robcazzaro , if you thinks my Gen2 style adc approach with loopFOC called by a TIMER callback might be a way to go, i can upload my code to a new github repo. That is something you could also work on with your blue pills :-)

robcazzaro, 2023-07-20:
@RoboDurden I can't promise I can do it in the next couple of days, but if you upload your code somewhere I'll have a look as soon as I have time.

RoboDurden, 2023-07-20:
Will then do so when i really take the break :-)
from the gd32 user manual: `ADC conversion time: 0.5μs for 12-bit resolution (2MS/s),`
Don't think that we have to worry a lot about also doing adc_buffer.v_batt every 100 μs (10 kHz `motor.loopFOC()` )

I also read 

```
Start of the conversion can be initiated:
- By hardware triggers with configurable polarity (internal timer events from TIMER0,TIMER1, TIMER2 and TIMER14)
```

So i might be able to "attach" the adc conversion begin to TIMER2 and only use the TIMER2 callback to run `motor.loopFOC();`

But yes, if the 4 adc conversion will only take a few microseconds, the values will mostly be "quite old" when the 100 μs TIMER2 callback runs loopFOC.

Don't know if this will affect the bldc closed loop much.

robcazzaro, 2023-07-20:
The conversion time depends on the ADC settings, and can be much higher. From the manual, section 10.4.8  `Example: CK_ADC = 14MHz and sample time is 1.5 cycles, the total conversion time is “1.5+12.5” CK_ADC cycles, that means 1us.`. I haven't looked at your settings, but can be more than 1us. Not sure where you found that reference, but it's not in the official documentation https://www.gigadevice.com.cn/Public/Uploads/uploadfile/files/20230314/GD32F130xxDatasheetRev3.7.pdf and https://www.gigadevice.com.cn/Public/Uploads/uploadfile/files/20230209/GD32F1x0_User_Manual_EN_Rev3.6.pdf

If the ADC is actually using DMA, the conversion time is not taking away processor time, but can result in the current ADC being sampled at the wrong time, depending on the code sampling the current or the battery first

RoboDurden, 2023-07-21:
Okay, it is ugly but extern "`C" {...}` does the job.

I have kept all code in one main.cpp and uploaded the repo to https://github.com/RoboDurden/GD32_SimpleFOC/blob/main/src/main.cpp

With the conversion defines of Gen2 i get this output when cutting off the the voltage:

```
OUT2T(fVoltage ,fCurrent)   OUTN(iCount)
25.06: -0.20  298607
25.06: 0.00  298687
25.06: 0.00  298767
25.06: 0.20  298847
23.01: 0.00  298927
13.41: 0.00  299007
10.83: -0.20  299087
8.36: -0.40  299167
6.41: 0.00  299247
```

here the uint16 values:
```
1042: 1354  231253
1041: 1354  231333
1042: 1354  231413
1028: 1354  231493
999: 1353  231573
975: 1354  231653
948: 1354  231733
923: 1353  231813
898: 1353  231893
870: 1354  231973
839: 1354  232053
801: 1353  232133
763: 1353  232213
727: 1353  232293
686: 1353  232373
649: 1354  232453
619: 1353  232533
569: 1353  232613
537: 1353  232693
503: 1354  232773
473: 1353  232853
440: 1353  232933
413: 1353  233013
385: 1353  233093
356: 1354  233173
326: 1353  233253
274: 1353  233333
270: 1370  233413
```

There is also `ADC_init2()` code that i found in the net and which might do continious conversion. It is also working.
https://gitee.com/yhalin/gd32-e230/blob/master/code/simple_app/main_adjust_Joystick.c

So of course it now would be nice to measure the microseconds for `ADC_init()` from `timer_cb()` to `DMA_Channel0_IRQHandler(void)` and for `ADCInit2()` the microseconds between each  `ADC_CMP_IRQHandler(void)` calls.

And of course doing it the c++ style without that ugly `extern "C"`

But i have to force myself now to stop and enjoy the rest of this day with packing all i need into my little camper.
But yes, i will take my test setups with me, at least the 50 km to my train station.

So @robcazzaro maybe you want to take a look at my low level code.
Doing the real LowsideTimerCurrentSense (derived from LowsideCurrentSense ) should not take much effort.

RoboDurden, 2023-07-21:
Added the microseconds evaluation: https://github.com/RoboDurden/GD32_SimpleFOC/blob/main/src/main.cpp#L48 
With the Gen2 style ADC_init() the two adc conversions finish 7-8 microseconds after the Timer2 handler triggered a new conversion.  As i plan to call motor.loopFOC only every 125 microseconds (= 8 kHz) this is absolutely fine :-)

RoboDurden, 2023-07-25:
Already spent whole sunday to port the working "bulk adc" to a new class `LowsideCurrentSenseTimer` but am having trouble with lots of statics needed to make the non c++ interrupt code work.
8 kHz Timer is running (orange led blinking) and triggering the bulk adc. Again after about 8 ms, the `DMA_Channel0_IRQHandler` gets called (green led blinking) :-)
But all adc values get written by dma as 65535 :-(

So directly porting the working https://github.com/RoboDurden/GD32_SimpleFOC/blob/main/src/main.cpp into a c++ class was a too big step :-/

Will first have to simply copy the `RCU_init(); DMA_init(); ADC_init();`	into the main.cpp and get valid adc values from there..

With LowsideCurrentSenseTimer we wil be able to have multiple motors and the option to add additional adc pins:
```

LowsideCurrentSenseTimer current_sense = LowsideCurrentSenseTimer(BLDC_CUR_Rds, BLDC_CUR_Gain, BLDC_CUR_G_PIN, BLDC_CUR_B_PIN, BLDC_CUR_Y_PIN);
LowsideCurrentSenseTimer current_senseLeft = LowsideCurrentSenseTimer(BLDC_CUR_Rds, BLDC_CUR_Gain, BLDC_CUR_L_G_PIN, BLDC_CUR_L_B_PIN, BLDC_CUR_L_Y_PIN);

setup()
{
  current_sense.AddAdc(VBATT_PIN,VBATT_CONVERT);                // add additional adc to bulk conversion
  current_sense.AddAdc(CURRENT_DC_PIN,CURRENT_DC_CONVERT,1000); // add with offest calculation with 1000 first values
  current_sense.AddCurrentSensor(current_senseLeft);          // add pinA, pinB and pinC to the one bulk conversion

if (current_sense.init(TIMER2,8000))    // user TIMER2 with 8 kHz
  {
    motor.linkCurrentSense(&current_sense);
    motorLeft.linkCurrentSense(&current_senseLeft); // motorLeft will only see current_senseLeft which will fetch its adc from master
```

code is online here
https://github.com/RoboDurden/Split_Hoverboard_SimpleFOC/blob/main/src/main.cpp#L46C1-L46C26
https://github.com/RoboDurden/Arduino-FOC/blob/master/src/current_sense/LowsideCurrentSenseTimer.cpp

ideas welcome :-/

Candas1, 2023-07-25:
You have limited time to sample the phase currents, and you have to do it at the right time.
This is why it's good practice to use inserted/injected adc.
Now with this bulk of adc channels to sample, your low side mosfets will have to stay ON for a longer time, so you will have to reduce the maximum duty cycle.
And I am not sure both motors will sample at the right moment.

That's why I said I can tackle the current sensing myself later.

RoboDurden, 2023-07-25:
**Ah now i understand why you always onsist on 'inserted' = 'synced' !** 
Adc of the low-side-currents of course need the low-side mosfets be ON for that time. Therefore the adc has to be in sync. 
If the ON phase of all three low-side mosfets would always start with the begin of a new TIMER0 cycle, then my bulk-method might still work and i would need the Gen2 code to trigger on TIMER0 to work.
Does the low-side pwm always begin with the ON-phase ?

like: 
```
|---_______|---_______|---_______| = 30% pwm ratio lo-side A
|-_________|-_________|-_________| = 10% pwm ratio lo-side B
|-----_____|-----_____|-----_____| = 50% pwm ratio lo-side C
```
 
However, over the 50% of the low-side C being on, the current might also differ, because the three hi-side mosfets  have different ON-times !

So what kind of low-side currents does the foc alghorithm needs anyway ? An average over the complete cycle ?
sorry yes, i should finish reading that long FOC intro article :-/

Using TIMER2 will not work my way.
Thanks for the clarification @Candas1  !!

Candas1, 2023-07-25:
Yes check that document I shared in the past, I can share the link again if needed.
It's related to Eferu's firmware but it's still relevant.

RoboDurden, 2023-07-25:
ha ha yes please post the link again. our threads here are so long that i have diffucultis to find something again.

Candas1, 2023-07-25:
[Here](https://docs.google.com/presentation/d/1pdJqmeX-wLChsHSGvrFmTE76WcokVxpkw2e2r_Hd5YY/edit?usp=sharing) it is

RoboDurden, 2023-07-25:
Thank @Candas1 you for the link (i meant that general introduction into FOC but this link is better suited for our low-level talk).
Now i think that "injected adc" is not about syncing the adc to the pwm but "Values must be read from registers instead of DMA.". 
And if i would trigger the adc by the TIMER0-callback (as is done in Gen2) AND would trigger that callback to the middle of the pwm cycle (`timerBldc_parameter_struct.alignedmode = TIMER_COUNTER_CENTER_DOWN;` ?), then i would have the same as the EFeru timing:
[image]
If it is not possible to set a custom counter value for TIMER0 to trigger the Timer0-callback, then i could use an additional TIMER2 to delay the adc-begin: Timer0 -> Timer2 -> ADC

Yes, adding VBATT and CURRENT_DC to the "bulk conversion" would unnessearily delay the end of conversion. But that would only be optional anyway.

What i still do not understand is what good it is to measure the currents when all three low-side mosfet are on and all highside-mofset ar off. Is the foc-alghorithmn only interessted in some inductive currents ?
But i do not really want to understand the physics of FOC anyway..

So if i succeed with syncing my adc-begin to the middle (plus some offset?) of TIMER0 (and somehow skip the DMA but read the values from some register to further shorten conversion time), then my code would be okay ?

Candas1, 2023-07-25:
The advantage of injected adc is that you can sample separately from the regular adc channels, it will just put any regular adc sampling on hold and start the injected adc instead.
In the gd32 drivers I already initialized the TIMER_CH_3 of TIMER0 for sampling.

What the gen2.0 firmware is doing with the adc is really wrong, that's not a good example.
It's starting an interrupt at Timer update event (middle), starting the adc by software in that interrupt instead of trigger it by the event, and then DMA is triggering another interrupt.

RoboDurden, 2023-07-25:
Okay but if you would tell me how to change Timer0 update event (middle) to update event (begin/end) then this awkward Timer0_cb->DMA_ADC->AdcFinished_cb should work.
I guess we could live a few micro seconds delay.

Timer0 update callback is initialized here (i guess):

Candas1, 2023-07-25:
we discussed some of it [here](https://github.com/Candas1/Split_Hoverboard_SimpleFOC/issues/4#issuecomment-1598203352)

If you are not interested in the physics of FOC or dealing with low level, forget about current sensing.

RoboDurden, 2023-07-25:
From the user manual:

**Center-aligned counting mode**
In the center-aligned counting mode, the counter counts up from 0 to the counter-reload value
and then counts down to 0 alternatively. The Timer module generates an overflow event when
the counter counts to the counter-reload value subtract 1 in the up-counting mode and
generates an underflow event when the counter counts to 1 in the down-counting mode. The
counting direction bit DIR in the TIMER0_CTL0 register is read-only and indicates the
counting direction when in the center-aligned mode. The counting direction is updated by
hardware automatically

[image]

Now It seems to me that the update event is fired in the middle (overflow) **AND** the begin/end (underflow)
So i would only (somehow) need to start adc conversion when DIR bit indicates an underflow event :-)

RoboDurden, 2023-07-25:
As a physicist i love abstractation - and strongly dislike all these Linux believers with their mindsets stuck in the 90s of last centrury who think that you first have to understand the complete set of underlying low levels to advance to a higher level :-(

This project here is about making the c++ simpleFOC library run on a GD32. 
And if that library wants the adc currents when all lo-side mosfets are on (and all hi-side mosfett are off), then i do **proudly** not want to understand why, because that belongs to a lower level.

0: physics level :-(
1: c-code GD32 level :-/
2: simpleFOC c++ level :-)

Candas1, 2023-07-25:
> This project here is about making the c++ simpleFOC library run on a GD32. And if that library wants the adc currents when all lo-side mosfets are on (and all hi-side mosfett are off), then i do **proudly** not want to understand why, because that belongs to a lower level.
That's still low level.
So you get to pick what level is interesting for you and ask when it's too level for you liking.

RoboDurden, 2023-07-25:
Exactly Candas :-)
When some issue to too low level (like the flag to only fire the updae event on underflow) to my liking i happly say it here and maybe you or @robcazzaro are happy to help because you low-level experts immediately know the answer.
When you don't answer, i might still find the happiness to dig deeper myself.

Do not forget that i only began here because you answered "There is still a lot for work to be done on the 6PWM driver before starting to work on current sensing."

Candas1, 2023-07-25:
I never suggested you should work on the current sensing.
Anyway, that's why I said feel free to experiment if you want to learn.
But if we have to step in all the time, that's not the most efficient use of our time.

Now I am discussing the extrapolation from the dev branch with the simplefoc team because you didn't want to discuss it with them. You could have handled this much better than me.

I repeat myself but [here](https://docs.google.com/spreadsheets/d/15msbDAIMxC2rIkq8Au8vf82ub1qEaS67Lc1cFb86Jpc/edit?pli=1#gid=654766453) I try to keep track of what I do and the information I collect.

Candas1, 2023-07-25:
Some signs were wrong in my case:
[image]

The extrapolation seems to work at constant speed at least now.
[image]

Need to do more tests.

RoboDurden, 2023-07-26:
Okay, 4x adc with TIMER2 working:
[image]
thanks @Candas1 for showing me StmStudio !
ai_AdcBuffer[0] and ai_AdcBuffer[1] (StmStudio can not import arrays ?) are the two low-side mosfet currents and i spin the motor by hand :-)
ai_AdcBuffer[2] is VBATT and changes nicely when i change the power supply voltage.
ai_AdcBuffer[3] is the overall current_dc but not yet calibrated.
Conversion time mostly takes about 14 micro seconds but sometimes up to 50 us:
[image]

But of course now i need to move back from TIMER2 to the pwm TIMER0. 
I dream that this `extern"C"{}` this time will make the Timer0 Update handler work.

P.S. how to also get the channel from Arduino-Pin:

```
		PinName pinname = DIGITAL_TO_PINNAME(aiPin[i]);
		uint32_t gd_pin =  gpio_pin[GD_PIN_GET(pinname)];
		uint32_t gd_port = gpio_port[GD_PORT_GET(pinname)];
		uint8_t gd_channel = get_adc_channel(pinname);
```

I simply wrote `analogRead(VBATT_PIN)` into main.cpp and did hit `F12` to find the c-level and the mapping macros/functions.

Candas1, 2023-07-26:
> StmStudio can not import arrays ?
When selecting the variables, there is an option to expend the arrays.

> P.S. how to also get the channel from Arduino-Pin:
>
> ```
> 		PinName pinname = DIGITAL_TO_PINNAME(aiPin[i]);
> 		uint32_t gd_pin =  gpio_pin[GD_PIN_GET(pinname)];
> 		uint32_t gd_port = [...]

Thanks. But ADC channel is OK, I am missing a function/macro to find the Timer channels for a pin.

RoboDurden, 2023-07-26:
my analog buffer is a dynamic array 
```

.h
uint16_t* LowsideCurrentSenseTimer::ai_AdcBuffer;
.cpp
ai_AdcBuffer = new uint16_t[aiPin.size()];
```
and only the constant pointer shows up in StmStudio

This gets expanded to 8 values:   `static uint16_t ai_AdcBufferLog[8];` :-/

i found this with timer.c

```
	// Brushless Control DC (BLDC) defines
	// Channel G
	#define BLDC_GH_PIN PA10
	#define BLDC_GL_PIN PB15
	// Channel B
	#define BLDC_BH_PIN PA9
	#define BLDC_BL_PIN PB14
	// Channel Y
	#define BLDC_YH_PIN PA8
	#define BLDC_YL_PIN PB13

	pwmDevice_t oPwmDevice = getTimerDeviceFromPinname(DIGITAL_TO_PINNAME(BLDC_GH_PIN));	// -> 2 = TIMER_CH_2
	SIMPLEFOC_DEBUG("Timer_CH ", oPwmDevice.channel);
	oPwmDevice = getTimerDeviceFromPinname(DIGITAL_TO_PINNAME(BLDC_GL_PIN));	// -> 1 = TIMER_CH_1
	SIMPLEFOC_DEBUG("Timer_CH ", oPwmDevice.channel);
	oPwmDevice = getTimerDeviceFromPinname(DIGITAL_TO_PINNAME(BLDC_BH_PIN));	// -> 1 = TIMER_CH_1
	SIMPLEFOC_DEBUG("Timer_CH ", oPwmDevice.channel);
	oPwmDevice = getTimerDeviceFromPinname(DIGITAL_TO_PINNAME(BLDC_BL_PIN));	// -> 0 = TIMER_CH_0
	SIMPLEFOC_DEBUG("Timer_CH ", oPwmDevice.channel);
	oPwmDevice = getTimerDeviceFromPinname(DIGITAL_TO_PINNAME(BLDC_YH_PIN));	// -> 0 = TIMER_CH_0
	SIMPLEFOC_DEBUG("Timer_CH ", oPwmDevice.channel);
	oPwmDevice = getTimerDeviceFromPinname(DIGITAL_TO_PINNAME(BLDC_YL_PIN));	// -> 0 = TIMER_CH_0
	SIMPLEFOC_DEBUG("Timer_CH ", oPwmDevice.channel);
```

But the channels seem to be wrongly shifted by 1 :-/

RoboDurden, 2023-07-26:
@Candas1, with `analogWrite(BLDC_GH_PIN,100);` and F12  i found this in `arduino/pwm.cpp`

```
void analogWrite(pin_size_t ulPin, int ulValue)
{
...
  set_pwm_value_with_base_period(ulPin, analogOut_period_us, value);

->

void set_pwm_value_with_base_period(pin_size_t ulPin, uint32_t base_period_us, uint32_t value)
{
...
    PWM pwm(ulPin);

->
PWM::PWM(uint32_t pin)
{
    PinName instance = DIGITAL_TO_PINNAME(pin);
..
    this->pwmDevice   = getTimerDeviceFromPinname(instance);
```

So the functions seem to be correct and my defines_2.0.h definitions wrong ?

Candas1, 2023-07-26:
I will check that getTimerDeviceFromPinname, but the GD_PIN_CHANNEL_GET macro it uses is also used in [get_adc_channel](https://github.com/CommunityGD32Cores/ArduinoCore-GD32/blob/dfa7ccbef5161f0215d28d9b59aa9f24e61d07f5/cores/arduino/analog.cpp#L277).
I might reach out to the arduino-gd32 team for clarification.
Thank you.

RoboDurden, 2023-07-26:
Yes, but in get_adc_channel with the line before it is 

```
  uint32_t function = pinmap_function(pinname, PinMap_ADC);
  uint32_t channel = GD_PIN_CHANNEL_GET(function);

```
whereas in getTimerDeviceFromPinname it is

```
  pinfunction = pinmap_function(instance, PinMap_PWM);
  pwmDevice.channel = GD_PIN_CHANNEL_GET(pinfunction);

```

Candas1, 2023-08-01:
Thanks, now I understand how it works.
I am waiting for possible fixes in [arduino-gd32](https://github.com/CommunityGD32Cores/ArduinoCore-GD32/issues/95#issuecomment-1657883059)

I started to work on the current sensing last evening.

[This](https://github.com/simplefoc/Arduino-FOC/issues/298) was missing in your splitboard code.

RoboDurden, 2023-08-01:
Good to hear from you Candas :-)

Yes, the current_sense.linkDriver(&driver); was missing in my splitboard main.cpp.
And yes: > Simplefoc is very well documented but this kind of things happens, and can generate unnecessary discussions in the forum.
I am the copy+paste guy who never reads documentations ;-)
Again the difference to the linux believers.. i have this expectation that code should be intuitive, speaking for itself and NOT need lengthy documentations..

As i have never reached the point where my current_sensor might be ready for testing with simpleFOC, i have never tested my current_sense.XY lines ;-)

I have nearly finished my mppt project and would like to return to this repo here.

Would be nice if you would also update this Split_Hoverboard_SimpleFOC and not only the Arduino-FOC (and the arduino-foc-drivers ) so i can see how you implement the SmoothedSensor.

Greetings from rainy Germany - nature loves rain :-)

Candas1, 2023-08-01:
As I told you I have a dev branch of this repository. It's up to date.

RoboDurden, 2023-08-01:
Okay now i see that you indeed have dev branches for all three github repos. Sorry, Github is not intutive for me ;-)

Candas1, 2023-08-01:
No worries.

RoboDurden, 2023-08-01:
Yes i think i also observed strange behavior with the SELF_HOLD_PIN / BUTTON_PIN as i could not reproduce the normal hoverboard behavior to turn off the board. Do not understand this.

To my understanding, the update event can trigger on overflow (in the middle) and underflow (beginning/ending).
I think the beginning/ending is the time when the adc should take place as the gate-high-states are centered to the middle and you said that measurenments should take place when the low-side mosfets are conduction = hi=off & lo=on

As i wanted to call an update handler to trigger adc, i think i can check if it was an underflow and only then start adc.
But you want to trigger adc directly i think. But i fear, the GD32 can not be configured to only fire the update event on underlow. Only overflow or overflow+underflow.
Then i guess you indeed need the channel3.
But with the channel3 you could also start the adc a few microseconds before the beginning/ending :-)

P.S. i had the idea to first do the time critical adc of gate currents 
then reconfigure the adc

```
	adc_channel_length_config(ADC_REGULAR_CHANNEL, aiPin.size());
	//adc_channel_length_config(ADC_REGULAR_CHANNEL, 1);

	int iSize = aiPin.size();
	for (int i=0; i<iSize; i++)
	{
		PinName pinname = DIGITAL_TO_PINNAME(aiPin[i]);
		uint32_t gd_pin =  gpio_pin[GD_PIN_GET(pinname)];
		uint32_t gd_port = gpio_port[GD_PORT_GET(pinname)];
		uint8_t gd_channel = get_adc_channel(pinname);
	
		gpio_mode_set(gd_port, GPIO_MODE_ANALOG, GPIO_PUPD_NONE, gd_pin);
		adc_regular_channel_config(i, gd_channel , ADC_SAMPLETIME_13POINT5);
	}

```
do the voltage and current_dc when we have all the time in the world
and finally reconfigure adc back to the two time critical gate currencies.

RoboDurden, 2023-08-01:
So you can configure this "inserted adc" only once at the beginning and the update/channel3 event directly triggers putting normal adc on hold and "insert" its adc 
and wait the few us to finish to read the values or only these inserted adc conversion gets written via dma ?
Everything without a handler function to be called ? 
Only (please) when these time critical adc have been finished to i can put a motor.loopFOC() in that final callback :-)

Candas1, 2023-08-01:
Yes regular adc is put on hold.
We need to check how analog read is implemented. I think Arduino-FOC-drivers has a voltage reading 'driver' that uses it. 

No 16khz is too fast for loopfoc.

RoboDurden, 2023-08-01:
If you give a callback when this inserted adc has finished, i can call loopFOC only every second call :-)

I overwrite the analog read function in my `LowsideCurrentSenseTimer` class

```
#define _ADC_VOLTAGE 3.3f
#define _ADC_RESOLUTION 4096.0f

float LowsideCurrentSenseTimer::ReadAdcVoltage(const int pin)
{
	if (_isset(pin))
		return ai_AdcBuffer[aiAdcMap[pin]] * _ADC_VOLTAGE / _ADC_RESOLUTION;
	return 0;
}
```

Candas1, 2023-08-01:
Got ch3 to work so I can define where I want to sample. It works without any interrupt.
I get one sinusoidale phase current for one pin, but the other one is not reacting, just noise.
Need to check if it's the right pin, if it's not defective, or if I did something wrong.

But I have guests from tomorrow to sunday so might be hard to find time to progress.

RoboDurden, 2023-08-02:
@Candas1 , i downloaded via zip file your `Split_Hoverboard_SimpleFOC-dev` but i get compiler error

`src\main.cpp:239:5: error: 'GPIO_BOP' was not declared in this scope; did you mean 'GPIOB'?
`
I also do not understand the syntax of 

```
  GPIO_BOP(GPIOB) = (uint32_t)GPIO_PIN_6; 
    motor.loopFOC();
    GPIO_BC(GPIOB) = (uint32_t)GPIO_PIN_6;
```
Are GPIO_BOP and GPIO_BC macro definitions ?

Also a lot warnings like

```
src\main.cpp:207:30: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  207 |   command.add('T', doTarget, "target voltage");
```

Am i missing something ?

P.S. with `command.add('T', doTarget, "target voltage");` i will be able to set the target speed to 10.2 Volt by entering `T10.2` via the RTTStream serial communication ?

RoboDurden, 2023-08-02:
@Candas1 , if i remove (comment) these strange GPIO_BOP macro lines and set target by

RoboDurden, 2023-08-02:
Also, the  `if (current_sense.init())` is still not active and i do not see a gd32 hardware file in https://github.com/Candas1/Arduino-FOC/tree/dev-gd32/src/current_sense/hardware_specific to "catch" the channel3 interrupt. Nor a new LowsideCurrentSensor class to implement your inserted adc - which of course i am eager to see :-)

ideas welcome.

RoboDurden, 2023-08-02:
Hi Candas :-)
As you can read i would be happy if you share your (complete) code that "Got ch3 to work".

RoboDurden, 2023-08-02:
But the timer0_update_handler is called exactly every 31 us:

```
	void TIMER0_BRK_UP_TRG_COM_IRQHandler(void)
	{

		if (SET == timer_interrupt_flag_get(TIMER_BLDC,TIMER_INT_FLAG_UP))
		{
			uint32_t iNow = getCurrentMicros();
			LowsideCurrentSenseTimer::iMicrosAdcReady = iNow - LowsideCurrentSenseTimer::iMicrosTimerCb;
				LowsideCurrentSenseTimer::iMicrosTimerCb = iNow;
			LowsideCurrentSenseTimer::iCountTimer0 = timer_counter_read(TIMER0);
```

Which reflects the 32 kHz (overflow + underflow) of the 16 kHz pwm.
But if the handler gets called exactly in sync, then the timer counter should always be the same (with some offset maybe). Yet i get

```
31: 80
31: 74
31: 1420
31: 79
31: 1475
31: 210
31: 430
31: 79 
```
(these log values are 100.000 microseconds appart.)

 `timer_counter_read(TIMER0);` might be wrong but it resolves to `TIMER_CNT(timer_periph)` which leads to 
`#define TIMER_CNT(timerx)                REG32((timerx) + 0x00000024U)         /*!< TIMER counter register */`
and the user manual confirms:
`This bit-filed indicates the current counter value. Writing to this bit-filed can change the value of the counter`

I do not understand this.

RoboDurden, 2023-08-02:
As the log output via RTT is only every 100000 micro seconds i have stored the last 10 calls 

```
			LowsideCurrentSenseTimer::aiMillis[LowsideCurrentSenseTimer::iPosMillis] = timer_counter_read(TIMER0);
			LowsideCurrentSenseTimer::iPosMillis = (LowsideCurrentSenseTimer::iPosMillis +1) % 10;

```
and that looks more like two different events within the 16 kHz period:
```
1413    1413    87      87      87      87      1412    1412    1412
87      87      1412    1412    1412    1412    88      88      88
87      87      1412    1412    1412    1412    1412    88      88
87      87      87      1412    1412    1412    1412    1412    88
87      87      87      1412    1412    1412    1412    88      88
1413    1413    87      87      87      87      1412    1412    1412
87      87      1412    1412    1412    1412    88      88      88
87      87      1412    1412    1412    1412    1412    88      88
87      87      87      1412    1412    1412    1412    88      88
87      87      87      1413    1413    1413    1413    88      88
1413    1413    87      87      87      87      1412    1412    1412
```

The range of the counter is 1500: `timerBldc_parameter_struct.period  = SystemCoreClock / pwm_frequency;`

But still i do not understand why 87 and 1412, and why they do not toggle like `87      1412      87      1412       87`

maybe you @robcazzaro  have an idea :-)

[83 more replies]