Candas1/Split_Hoverboard_SimpleFOC #8: adding Gen1 enviroment (STM32F103RC) ..

RoboDurden, 2023-07-10:
Hi @Candas1 , thank you for accepting my pull request.
As i thought you would not want to do that, i pushed more changes to my fork that now found it's way into your repo.

I added the STM32F103RC enviroment that i found in the EFeruFOC platformio.ini and the code compiles.
But the code seems to hang in setup()

`  oKeepOn.Set(true);  // now we can release the on button :-)
`
when i debug and step_over which does nothing but `digitalWrite(m_iPin,bOn);`

For that i also rearranged the defines_x-y.h in config.h and defines.h to allow for Gen1 and Gen2 boards.
You might like that.

```
#if defined(PLATFORMIO)
  #ifdef STM32F103RC
    #define HOVER_GEN   1
    #define HOVER_LAYOUT	0
  #else
    #define HOVER_GEN   2
    #define HOVER_LAYOUT	0
  #endif
#else
  // LAYOUT_x_y is used in defines.h
  #define HOVER_GEN   2
  #define HOVER_LAYOUT 0
  //  2_0	// https://github.com/flo199213/Hoverboard-Firmware-Hack-Gen2
  //  2_1	// https://github.com/krisstakos/Hoverboard-Firmware-Hack-Gen2.1
  //  2_2	// 2023/05/11 only MASTER and TEST_SPEED: motor is spinning but needs a push to startup :-/
  //  2_4	// NOT READY !!! https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/3
  //  2_5	// NOT READY !!! https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/11

  //  1_0	// old Gen 1 boards with two motors
#endif

```

In `main.c` there now is my test code to optimize `zero_electric_angle`

```
    OUT2T(fSpeedMin , fSpeedMax)
    OUT2T(fSpeedAvgMin , fSpeedAvgMax)
    OUT2T(abs(fSpeedMin+fSpeedMax), abs(fSpeedAvgMin+fSpeedAvgMax))
    //OUT2T("fLinAdd%",sensor.fLinAdd*100)
    OUT2T(fOptimize*1000, motor.zero_electric_angle*1000)
    OUT2T( iMicros , iMicrosMax )
```
You might want check your `2.09` again but feel free to delete all the `fSpeedXY `stuff :-)

the .gitignore now contains my failed attempt to push my HallSensor changes, you better remove that from your repo :-)

```
 # Ignore everything
*
# But descend into directories
!*/
# Recursively allow files under subtree
!.pio/libdeps/GD32F130C8/Simple FOC/src/**
```

When you test your Gen2 with a stm32 mcu, you might get stm32_env to work. 
I did not have any idea what i was doing when simply copy and paste the env from the EFeru platformio.ini

Candas1, 2023-07-10:
Hi,

As you insisted, I thought this was relevant and tested code.

I will need to clean it up now.

RoboDurden, 2023-07-10:
Well it was relevant and tested code when i made the pull request.
But as you said you would not accept it, i began to continue with my fork..
Luckily i have not pushed my recent testings for quite a while, so what you got is still quite good for your repo :-)

RoboDurden, 2023-07-10:
Dear @Candas1 , as you probably remember my rude words for nice Emanuel Feru, you should know that i for sure am not an easy character - my nick name (alter ego) Robo Durden is a tribute to Tyler Durden !!!
So for sure never do anything only because i "insist" on it.
And as i try my best to be a born loser ('it is only after we have lost everything that we are free to do anything" - Tyler Durden), i never really insist on anything :-)

RoboDurden, 2023-07-13:
thanks @robcazzaro for the reminder that Gen1 is stm32 and we do not need two motors with our Gen2 GD32 port.

I have made some progress with running simpleFOC on my Gen1 test setup:

```
MOT: Monitor enabled!  
Split Hoverboards with C++ SimpleFOC :-)  
driver.init() succeeded :-)  
MOT: Init  
MOT: Enable driver.  
MOT: Align sensor.  
MOT: Skip dir calib.  
MOT: Skip offset calib.  
MOT: No current sense.  
MOT: Ready.  
setup needed ms: 3308  
0  
TIME_SEND  
0  
0  
0  
0  
0  
0  
1  
1  
1  
1  
1  
1  
2  
2  
2  
2  
2  
3  
7
```

Don't know why  execution hangs/resets at

```
  oKeepOn.Init();
  oKeepOn.Set(true);  // now we can release the on button :-)
  oOnOff.Init();
```

But i for now have disabled that functionality by simply 
```
// Self hold defines
// Set output high to keep the latch ON. Set low to turn the board OFF
#define SELF_HOLD_PIN TODO_PIN	// PA5	// OFF_PIN
```

This is how i was able to activate UART3 debug output:

```
  #if HOVER_GEN == 1
    //HardwareSerial oSerialSteer(PA3, PA2);  // long cable NOT 5V tolerant EFeru USART2 GPIO Configuration
    HardwareSerial oSerialSteer(PB11, PB10);  // short cable 5VT EFeru USART3 GPIO Configuration
    #undef DEBUG_UART
    #define DEBUG_UART oSerialSteer
  #endif
```
https://github.com/RoboDurden/Split_Hoverboard_SimpleFOC/blob/main/src/main.cpp#L19

@Candas1 , if you have not yet worked on your repo since accepting my old pull request, you might want me to clean up my current fork and do another pull request.

But the simpleFOC does not yet work on the Gen1.
In the debug log you see the millis() of every loop(), so the code is running.
But quickly a constant  overcurrent of 1.6A (limit of 1.5A set with my little power supply), the motor makes a little buzzing noise and voltage drops from 25 to 21 Volt.
With a current limit of 1.0A, the mcu reboots.
Do not really want to increase the current limit. 1.5A should be well enough to make the motor startup.

I have to check the https://github.com/RoboDurden/Split_Hoverboard_SimpleFOC/blob/main/include/defines_1-0.h to make sure i got all pin assignments correct from the EFeru defines.h

Ideas welcome :-)

If you know someone with knowledge about Darrieus VAWT wind turbines: https://youtu.be/UitxRSDj3tc

RoboDurden, 2023-07-14:
```
SIMPLEFOC_PWM_LOWSIDE_ACTIVE_HIGH
sets PWM polarity on low side phase PWM outputs - default is active high (true) but can be set to false for active low polarity. Affects only low side FETs in 6-PWM mode. No effect in other modes.
SIMPLEFOC_PWM_HIGHSIDE_ACTIVE_HIGH
sets PWM polarity on high side phase PWM outputs - default is active high (true) but can be set to false for active low polarity. Affects only high side FETs in 6-PWM mode. No effect in other modes.
```

I remember that @robcazzaro mentioned in some of our long issues that the lo-side digital-out pin is negated ? But that would only apply to the Gen2 boards ?

In EFeru::setup.c i find

```
  sConfigOC.OCPolarity   = TIM_OCPOLARITY_HIGH;
  sConfigOC.OCNPolarity  = TIM_OCNPOLARITY_LOW;
```
and only three channels per motor.
```
  HAL_TIM_PWM_ConfigChannel(&htim_left, &sConfigOC, TIM_CHANNEL_1);
  HAL_TIM_PWM_ConfigChannel(&htim_left, &sConfigOC, TIM_CHANNEL_2);
  HAL_TIM_PWM_ConfigChannel(&htim_left, &sConfigOC, TIM_CHANNEL_3);
```
Maybe that is what simpleFOC is doing with BLDCDriver3PWM and high will open the hi-mosfet and low the low-side mosfet ?

then i should set that build flag SIMPLEFOC_PWM_LOWSIDE_ACTIVE_HIGH to false ?
After i have looked at the schematics pdf i can confirm that a low-side HIGH will close the the lo-side mosfet and a LOW will make it conducting.

Will try SIMPLEFOC_PWM_LOWSIDE_ACTIVE_HIGH = false , but i do not really know how to set build flags. And a flag normaly is only a flag and not a parameter..

RoboDurden, 2023-07-14:
Okay motor is spinng - at least in the right directions. But very poorly with buzzing noise and kind of very coarse steps. Right rotation is a bit more smooth and only draws about 300mA at 26V. opposite rotation about 400-500mA.

```
  driver.voltage_limit = 0.3 * driver.voltage_power_supply;   // 0.3 = keep well below 1.0 for testing !
  motor.voltage_limit = 0.3 * driver.voltage_limit; // stupid bug to have two voltage_limit in different places
```

Does someone on the simpleFOC forums already succeeded with Gen2 ?

I would like to have working Gen1 to test my HallSensor linear prediction..

RoboDurden, 2023-07-15:
Ah, i had to calibrate the motor:
```
MOT: Init
MOT: Enable driver.
MOT: Align sensor.
MOT: sensor_direction==CCW
MOT: PP check: fail - estimated pp: 18.00
MOT: Zero elec. angle: 4.19
MOT: No current sense.
MOT: Ready.
setup needed ms: 6428
```
Now it spins nicely. I have a different motor in the stm setup :-/